lardemua/atom #8: Sensors first guess's orientation isn't correctly saved

afonsocastro, 2019-05-12:
`        for joint in robot.joints:
            for sensor in robot.sensors:
                if sensor.parent == joint.child:
                    optimization_parent_link = joint.parent
            for mp in marker_poses:
                (trans, rot) = listener2.lookupTransform(optimization_parent_link, mp.child_frame_id, rospy.Time(0))
                if joint.child + "_first_guess" == mp.child_frame_id:
                    joint.origin.xyz[0] = trans[0]
                    joint.origin.xyz[1] = trans[1]
                    joint.origin.xyz[2] = trans[2]
                    joint.origin.rpy[0] = rot[0]
                    joint.origin.rpy[1] = rot[1]
                    joint.origin.rpy[2] = rot[2] `

miguelriemoliveira, 2019-05-12:
Hi @afonsocastro ,

try it out with my code, I think it is working ...

```bash
rosrun interactive_marker_test create_first_guess.py -w car_center
```

miguelriemoliveira, 2019-05-12:
And don't forget to assign someone to this issue as well ...

afonsocastro, 2019-05-12:
yap, it's working well with your code

closed 2019-05-12 by afonsocastro