lardemua/atom #15: Use PR2 for complete tf trees

miguelriemoliveira, 2019-05-13:


miguelriemoliveira, 2019-05-15:
Hi @afonsocastro , I am not sure it is very easy to collect data from a PR2 as was your task.

I will try to find a rosbag file of a real PR2 in my backups. Will get back to you on this...

afonsocastro, 2019-05-15:
I was trying to get one already. Actually, it's not the simplest thing... i will wait for that rosbag

afonsocastro, 2019-05-15:
Right now, i have this. It's needed to set 'false' the car_model parameter when we run the launch file in order to see the pr2 robot instead of the car model

miguelriemoliveira, 2019-05-15:
Great. I will search for the bag file and get back to you ...

afonsocastro, 2019-05-16:
When it comes to parse the PR2 xacro, there are some problems occurring as the 'type' attribute and the 'compensator' tag are unknown in some links. I'll review this later

miguelriemoliveira, 2019-05-16:
Not very important I think. I will create a low priority issue for this.

miguelriemoliveira, 2019-05-16:
Hi @afonsocastro . I email my friend. Let's see what he replies.

Also I found this. Can you check if you can find a bag file with multiple sensors?

http://projects.csail.mit.edu/stata/downloads.php

miguelriemoliveira, 2019-05-16:
I took a look and recomend this bag:

2012-01-25-12-14-25.bag

http://infinity.csail.mit.edu/data/2012/2012-01-25-12-14-25.bag

Should contain multiple cameeas and lasers.

```yaml
path: 2012-01-25-12-14-25.bag
version: 2.0
duration: 1130.292904
start: 1327522466.090780
end: 1327523596.383684
size: 52374992833
messages: 682837
indexed: True
compression: none
types:
    - type: geometry_msgs/PoseWithCovarianceStamped
      md5: 953b798c0f514ff060a53a3498ce6246
    - type: nav_msgs/Odometry
      md5: cd5e73d190d741a2f92e81eda573aca7
    - type: pr2_mechanism_controllers/BaseOdometryState
      md5: 8a483e137ebc37abafa4c26549091dd6
    - type: pr2_mechanism_controllers/Odometer
      md5: 1f1d53743f4592ee455aa3eaf9019457
    - type: sensor_msgs/CameraInfo
      md5: c9a58c1b0b154e0e6da7578cb991d214
    - type: sensor_msgs/Image
      md5: 060021388200f6f0f447d0fcd9c64743
    - type: sensor_msgs/Imu
      md5: 6a62c6daae103f4ff57a132d6f95cec2
    - type: sensor_msgs/LaserScan
      md5: 90c7ef2dc6895d81024acba2ac42f369
    - type: std_msgs/Bool
      md5: 8b94c1b53db61fb6aed406028ad6332a
    - type: tf/tfMessage
      md5: 94810edda583a504dfda3829e70d7eec
topics:
    - topic: /base_odometry/odom
      type: nav_msgs/Odometry
      messages: 104484
      frequency: 44150.5684
    - topic: /base_odometry/odometer
      type: pr2_mechanism_controllers/Odometer
      messages: 1130
      frequency: 1.0000
    - topic: /base_odometry/state
      type: pr2_mechanism_controllers/BaseOdometryState
      messages: 1130
      frequency: 1.0000
    - topic: /base_scan
      type: sensor_msgs/LaserScan
      messages: 22542
      frequency: 20.0033
    - topic: /camera/depth/camera_info
      type: sensor_msgs/CameraInfo
      messages: 33758
      frequency: 30.7120
    - topic: /camera/depth/image_raw
      type: sensor_msgs/Image
      messages: 33758
      frequency: 30.8725
    - topic: /camera/rgb/camera_info
      type: sensor_msgs/CameraInfo
      messages: 33790
      frequency: 31.1159
    - topic: /camera/rgb/image_raw
      type: sensor_msgs/Image
      messages: 33790
      frequency: 31.0505
    - topic: /robot_pose_ekf/odom_combined
      type: geometry_msgs/PoseWithCovarianceStamped
      messages: 32549
      frequency: 24.9826
    - topic: /tf
      type: tf/tfMessage
      messages: 117750
      connections: 2
      frequency: 556.0525
    - topic: /tilt_scan
      type: sensor_msgs/LaserScan
      messages: 22541
      frequency: 19.9918
    - topic: /torso_lift_imu/data
      type: sensor_msgs/Imu
      messages: 113020
      frequency: 100.0049
    - topic: /torso_lift_imu/is_calibrated
      type: std_msgs/Bool
      messages: 1
    - topic: /wide_stereo/left/camera_info
      type: sensor_msgs/CameraInfo
      messages: 33152
      frequency: 29.4106
    - topic: /wide_stereo/left/image_raw
      type: sensor_msgs/Image
      messages: 33152
      frequency: 29.4096
    - topic: /wide_stereo/right/camera_info
      type: sensor_msgs/CameraInfo
      messages: 33145
      frequency: 29.4102
    - topic: /wide_stereo/right/image_raw
      type: sensor_msgs/Image
      messages: 33145
      frequency: 29.4098
```

miguelriemoliveira, 2019-05-16:
It contains over 1000 secs and is a 40GB file. 

I have no space for this ... @afonsocastro can you download and create a smaller bag file, say with 1 or 2 minutes only?

miguelriemoliveira, 2019-05-16:
How to split a bag file:
https://answers.ros.org/question/99711/how-to-split-a-recorded-rosbag-file/

afonsocastro, 2019-05-17:
Yes, i'll do it

closed 2019-05-21 by miguelriemoliveira