lardemua/atom #18: Why the f*%$k is the car model misplaced?

miguelriemoliveira, 2019-05-15:
This occurs on my system but not in @afonsocastro 's system. They are git synced so I am really not sure what is happening.

@afonsocastro , can you test it on a third system to see if it is your system or mine that is out of the norm?

afonsocastro, 2019-05-15:
I've already found the problem! you need to put on your .bashrc file this command: 

export LC_NUMERIC="en_US.UTF-8"

It probably have to do with the comma/point system (american or european). I tried without this line on my .bashrc and the car model showed up misplaced too. This command is on Read.me already.

closed 2019-05-15 by afonsocastro

miguelriemoliveira, 2019-05-15:
Great. So what was needed was the expletive? It is amazing the amount of stuff that starts working once we start cursing :-) ...