lardemua/atom #22: Create or get a bag file of an autonomous vehicle (or Instrumented Car)

miguelriemoliveira, 2019-05-16:
Related to #21

miguelriemoliveira, 2019-05-16:
Found some bags fro mthe MIT autonomous car at lars.mec.ua.pt//home/lardata/bags/Thesis_mriem/sequence1.bag

miguelriemoliveira, 2019-05-16:
Another interesting dataset is 

https://epan-utbm.github.io/utbm_robocar_dataset/

miguelriemoliveira, 2019-05-17:
Concerning https://epan-utbm.github.io/utbm_robocar_dataset/ , I cannot find a bagfile with images. Only with lasers ...

@afonsocastro can you?

afonsocastro, 2019-05-17:
I have a bag with images. This bag has one camera and 7 lasers. 520 seconds of recording. The bag was recorded last year from the atlascar2

miguelriemoliveira, 2019-05-17:
Great! Please copy paste here the output of 

```bash
rosbag info
```

afonsocastro, 2019-05-17:
path:        Volta_UA.bag
version:     2.0
duration:    8:40s (520s)
start:       May 18 2018 17:23:22.46 (1526660602.46)
end:         May 18 2018 17:32:02.97 (1526661122.97)
size:        22.8 GB
messages:    643939
compression: none [5694/5694 chunks]
types:       diagnostic_msgs/DiagnosticArray          [60810da900de1dd6ddd437c3503511da]
             geometry_msgs/PoseStamped                [d3812c3cbc69362b77dc0b19b345f8f5]
             gps_common/GPSFix                        [3db3d0a7bc53054c67c528af84710b70]
             nav_msgs/Odometry                        [cd5e73d190d741a2f92e81eda573aca7]
             novatel_gps_msgs/Gpgga                   [8f51cb504898f39d8ad9f698f0b6f9cd]
             novatel_gps_msgs/Inscov                  [a4ae1586410fc24e8ab4019825bb8bdd]
             novatel_gps_msgs/Inspva                  [f6fbcfee08847158b28edeb7f76b942f]
             novatel_gps_msgs/NovatelCorrectedImuData [81ec3aad90f65315c03ad2199cdd99cf]
             novatel_gps_msgs/NovatelPosition         [16508625c306bc93a852c2381bb2573c]
             novatel_gps_msgs/NovatelVelocity         [a8fb7d9232aaf68f98caa2b4cda2597b]
             sensor_msgs/Image                        [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu                          [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/LaserScan                    [90c7ef2dc6895d81024acba2ac42f369]
             sensor_msgs/NavSatFix                    [2d3a8cd499b9b4a0249fb98fd05cfa48]
             sensor_msgs/PointCloud2                  [1158d486dd51d683ce2f1be655c3c181]
             std_msgs/Float32MultiArray               [6a40e0ffa6a17a503ac3f8616991b1f6]
             tf2_msgs/TFMessage                       [94810edda583a504dfda3829e70d7eec]
topics:      /DadosInclin           23552 msgs    : std_msgs/Float32MultiArray              
             /bestpos               10934 msgs    : novatel_gps_msgs/NovatelPosition        
             /bestvel               10414 msgs    : novatel_gps_msgs/NovatelVelocity        
             /camera/image_color     3645 msgs    : sensor_msgs/Image                       
             /cloud_Outliers         2073 msgs    : sensor_msgs/PointCloud2                 
             /cloud_Total            2073 msgs    : sensor_msgs/PointCloud2                 
             /cloud_minada3          2073 msgs    : sensor_msgs/PointCloud2                 
             /corrimudata           52067 msgs    : novatel_gps_msgs/NovatelCorrectedImuData
             /diagnostics            2052 msgs    : diagnostic_msgs/DiagnosticArray          (4 connections)
             /fix                   10411 msgs    : sensor_msgs/NavSatFix                   
             /gpgga                 10413 msgs    : novatel_gps_msgs/Gpgga                  
             /imu                   52065 msgs    : sensor_msgs/Imu                         
             /inscov                  520 msgs    : novatel_gps_msgs/Inscov                 
             /inspva                52065 msgs    : novatel_gps_msgs/Inspva                 
             /ld_rms/scan0          26032 msgs    : sensor_msgs/LaserScan                   
             /ld_rms/scan1          26031 msgs    : sensor_msgs/LaserScan                   
             /ld_rms/scan2          26027 msgs    : sensor_msgs/LaserScan                   
             /ld_rms/scan3          26027 msgs    : sensor_msgs/LaserScan                   
             /lms151_D_scan         25938 msgs    : sensor_msgs/LaserScan                   
             /lms151_E_scan         25972 msgs    : sensor_msgs/LaserScan                   
             /odom                  10408 msgs    : nav_msgs/Odometry                       
             /tf                   232738 msgs    : tf2_msgs/TFMessage                       (6 connections)
             gps                    10408 msgs    : gps_common/GPSFix                       
             local_xy_origin            1 msg     : geometry_msgs/PoseStamped

afonsocastro, 2019-05-17:
[image]
[image]

afonsocastro, 2019-05-17:
I think this is what we need. i have some others bags too, with less recording duration

miguelriemoliveira, 2019-05-17:
Great. We can start working on this one.

When you copy better to copy the text only. See [this example](https://github.com/lardemua/AtlasCarCalibration/issues/15#issuecomment-493252565) .

We do not need a bag file with a large duration. Just a minute would be ok. Can you crop the bagfile into a smaller one that can be shared around?

miguelriemoliveira, 2019-05-21:
Hi @afonsocastro  ,

Tiago Crispin (@hellxberg ) also needs the bag file you have.

Can you give it to him. 

Also, for me it would be interesting that we have a smaller demo version of this bag file. Small enough to upload to github so everyone could test with the demo bag file.

Can you crop just a minute or two and upload it? I will explain Tiago how this is done and then he can help you if you need.

miguelriemoliveira, 2019-05-27:
Got one today with the atlascar2. Also, it contains views of chessboards.

miguelriemoliveira, 2019-05-27:
There is a problem with the camera_info message. Will close this one. Continue on #31

closed 2019-05-27 by miguelriemoliveira