lardemua/atom #26: rviz crashes when changing desktops

miguelriemoliveira, 2019-05-21:
When running 

```bash
roslaunch interactive_marker_test pr2_calibration.launch
```

whenever I change desktops rviz crashes with the following error:

```bash
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
================================================================================REQUIRED process [rviz-3] has died!
process has died [pid 13431, exit code -6, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/mike/catkin_ws/src/AtlasCarCalibration/interactive_marker_test/calibrations/pr2/config.rviz __name:=rviz __log:=/home/mike/.ros/log/c227165a-7bcb-11e9-ba2d-0028f80a63ec/rviz-3.log].
log file: /home/mike/.ros/log/c227165a-7bcb-11e9-ba2d-0028f80a63ec/rviz-3*.log
Initiating shutdown!
================================================================================
```

miguelriemoliveira, 2019-05-21:
Hi @tiagomfmadeira ,

I remember you had the same problem. Did you find out how to solve it?

miguelriemoliveira, 2019-05-21:
well I found out that this is related to the camera marker...

miguelriemoliveira, 2019-05-21:
Moreover, it happens only if I am running a rosbag playback ... weird.

Anyway, I will use an Image marker instead. For now it should suffice.

Will close but @tiagomfmadeira if you have any good ideas please tell me.

closed 2019-05-21 by miguelriemoliveira

tiagomfmadeira, 2019-05-21:
Hello @miguelriemoliveira,

Can confirm this is also a problem in elementary OS 5 Juno. Perhaps because it's based on Ubuntu 18.04.2 LTS (Bionic Beaver).

Did not find a solution for it unfortunately.

miguelriemoliveira, 2019-05-21:
OK, thanks!