lardemua/atom #31: New atlascar bag file contains incorrect camera_info message

miguelriemoliveira, 2019-05-27:
I guess we need a new bag file.

miguelriemoliveira, 2019-05-27:
The message contains all zero values as follows:

```bash
header: 
  seq: 852
  stamp: 
    secs: 1558955393
    nsecs: 509442614
  frame_id: "frontal_camera"
height: 0
width: 0
distortion_model: ''
D: []
K: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
binning_x: 1
binning_y: 1
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: True

```

miguelriemoliveira, 2019-05-27:
Before the new bag file we must calibrate the camera (all cameras if possible, if not the frontal one).

Created #32 for that.

miguelriemoliveira, 2019-05-27:
So far, incorrect stuff is:

1. camera_info message contains zero value
2. no left and right laser scans
3. image from frontal camera is very slow rate and seems like it is not debayered

miguelriemoliveira, 2019-05-27:
@afonsocastro I told this never works on the first attempt :). Lets try it again on Thursday, 9h30?

afonsocastro, 2019-05-29:
we can try again with atlascar, or we can use the bag that i have. I will crop the bagfile

miguelriemoliveira, 2019-05-29:
The bag file you have does not have chessboards, so it does not work.

miguelriemoliveira, 2019-06-07:
Now we have a nice bag file with everything we need for now. Closing.

closed 2019-06-07 by miguelriemoliveira