lardemua/atom #32: Calibrate frontal camera

miguelriemoliveira, 2019-05-27:
follow this 

http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

miguelriemoliveira, 2019-05-27:
@tmralmeida can you do this with @afonsocastro ? Do you have experience with this?

miguelriemoliveira, 2019-05-28:
Hi @hellxberg do you have the yaml file for the front camera ? You used the intrinsic value did you not?

@tmralmeida or @bernardomig did you backup the disk on the computer when you installed ubuntu 18.04?

hellxberg, 2019-05-29:
Hi @miguelriemoliveira, I did use intrinsic value of the camera but i didn't use any yaml file. I recorded all the information i have with "rosbag play".

miguelriemoliveira, 2019-05-29:
Ok, can you please copy paste here the result of a rostopic echo /.../camera_info

miguelriemoliveira, 2019-05-30:
Here are the contents of a new calibration yaml.

```yaml
image_width: 1624
image_height: 1224
camera_name: 14233704
camera_matrix:
  rows: 3
  cols: 3
  data: [1420.060841543088, 0, 794.7046593573046, 0, 1421.214911871335, 564.6352822046543, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.2086707440041696, 0.1741483019004053, -0.002057879830518041, -0.0002302521057234884, 0]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [1344.62646484375, 0, 793.1936514067638, 0, 0, 1370.65185546875, 559.9345065366651, 0, 0, 0, 1, 0]
```

closed 2019-05-30 by afonsocastro