lardemua/atom #35: Lidar data labelling

afonsocastro, 2019-09-28:
Hi @miguelriemoliveira ,
I tested to label the lidar data as you asked and it worked! For now, the labeller allows NaN in range measurements, but we have the problem of only recording some ranges between all the important points. More than that, the size of the cluster collected change in every capture and for each laser...

miguelriemoliveira, 2019-09-28:
Hi @afonsocastro 

yes, the size of the clusters will change and the measurements are irregular. 

That could only be changed with a new calibration pattern (non reflective paper?).

However, remember that our best guess for a LIDAR data cost function was to use only the extrema points  of the cluster. Since our pattern has a white canvas, we should always have measurements in those regions.

I will try to implement a cost function using these lidar data a the new dataset (#36 ). Lets see if we get better results then...

miguelriemoliveira, 2019-11-08:
This is done as well, right?

closed 2019-11-08 by miguelriemoliveira