lardemua/atom #37: Sensors Pose First Guess not loaded

afonsocastro, 2019-09-30:
If we launch _roslaunch interactive_calibration atlascar2_calibration.launch_ and then _rosrun interactive_calibration create_first_guess.py -w base_link_, we can move the sensors freely and see directly the changes of the point clouds and the images (as it was expected).

Then we can save our best eye-belief config for the sensors pose. The problem is when I try to re-run the launch file with the read_first_guess argument, rviz opens and the sensors pose are exactly the same as in the original urdf robot description. 
I found that the problem is related to the bag file that is running because if I test this first guess process without running any bag file it all goes well...
I think that the bag must be publishing some tf's about the sensors pose, but I am not sure of this. @miguelriemoliveira, can you help me figure out what is wrong here?

miguelriemoliveira, 2019-09-30:
I will look into it right now ...

afonsocastro, 2019-09-30:
I have a tip that might be related:
I was trying to get some calibration (collecting some snapshots and then optimize the pose), but I realized that the json file created by the data_collector_and_labeler script didn't store any transforms... This means that there were no transforms being published by the bag file, am I right?
That's why I think it's all related to this new bag

miguelriemoliveira, 2019-09-30:
Sorry, had a phone call until now. I will look into it now ...

miguelriemoliveira, 2019-09-30:
Not sure if I will have time to work on this tonight. If not sooner
tomorrow...

miguelriemoliveira, 2019-10-01:
Hi @afonsocastro ,

Sorry for the delay. It was a crazy day. I am working on this now.

this is very strange. The first time I tested your procedure it worked!!! The sensor poses were saved.

Then I did it a second time and ... does not work!!!

Will keep looking into it.

miguelriemoliveira, 2019-10-01:
Hi @afonsocastro ,

got it. This one was not easy, there was a tf_static topic on the bag file (should it be there, not sure) which overlapped (only sometimes) the tf_static published by the robot_state_publisher (using the first guess file).

I had to remap the tf_static topics in the bag file when playing to avoid the conflict.

```xml
<node pkg="rosbag" type="play" name="rosbag_play" output="screen"
          args=" $(arg bag) --clock -r $(arg rate) -l -s 0 /tf:=/tf_dev_null /tf_static:=/tf_static_dev_null"/>
```

You must now indicate the path to the "first guess" file so that the script knows were to save. For example: 

```bash
rosrun interactive_calibration create_first_guess.py -w base_link -f /calibrations/atlascar2/first_guess.urdf.xacro
```

The launch file already reads this one by default, but you can change the "xacro_file_first_guess" argument if you will.

The readme was updated accordinlgy.

Please test and let me know if it works.

afonsocastro, 2019-10-02:
Hi! 
well, it was, as I thought, about the published topics... But I really needed your help :) 
It all works just fine now, thank you @miguelriemoliveira.
I will have some results and conclusions soon, for this bag

miguelriemoliveira, 2019-10-02:
Great. We should meet next week for discussing you writting the thesis. Perhaps Tuesday, take it as the deadline for having some results. Then you **must** start writting. We can always continue the work after the thesis is written.

afonsocastro, 2019-10-02:
Ok, next Tuesday I can bring a skeleton of the thesis with some topics of what I think it should be said in each section of the document. That and some camera-to-camera calibration results.

miguelriemoliveira, 2019-10-02:
Great!

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

afonsocastro, 2019-10-08:
Hi @miguelriemoliveira,
my thesis skeleton is ready. We will talk about it tomorrow, as well?

miguelriemoliveira, 2019-10-08:
Yes. we must.

Miguel
*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2019-11-08:
Hi @afonsocastro , if this is solved please close.

commit 0e3e9c49 (2019-10-01): fix #37 . This was a tricky one: there were some static tf messages in the bag file.
closed 2019-11-08 by afonsocastro