lardemua/atom #42: Collect a bag file with the UR10 for calibrating camera w.r.t. robot

miguelriemoliveira, 2019-11-05:


eupedrosa, 2019-11-08:
First bag created. Here is a single image.

[image]

Do you think that this kinda of data will be enough? The light conditions are not the best.

miguelriemoliveira, 2019-11-08:
It looks good. I think it looks fine, but the only way to know is to see if the checkerboard is detected in the images.

We use this line to perform the detection:

https://github.com/lardemua/AtlasCarCalibration/blob/a652897ec55d63126fabc8a13d822cef120f0870/interactive_calibration/src/interactive_calibration/interactive_data_labeler.py#L231

Perhaps you can create a simple test script with receives an image  filename as input  (or a topic name) and shows the image with the detection overlayed? ...

If you want I can try to do that ...

Did you start with a description of the robot already? You could add a folder in https://github.com/lardemua/AtlasCarCalibration/tree/master/interactive_calibration/calibrations
with "your robot".

eupedrosa, 2019-11-08:
> Perhaps you can create a simple test script with receives an image filename as input (or a topic name) and shows the image with the detection overlayed? ...

I will do this!! I will add the script to the interaction_calibration for future use.
> Did you start with a description of the robot already?

Yes!!. And I will add the xacros to "my robot".

miguelriemoliveira, 2019-11-08:
Great. Push when you can. I am eager to take a look!

eupedrosa, 2019-11-08:
Here we have good example:
[image]

And here we have a bad example:
[image]

I think that I'm going to create a new bag with this utility window opened :smiley:

miguelriemoliveira, 2019-11-08:
the utility is already becoming useful :)

miguelriemoliveira, 2019-12-02:
Hi @eupedrosa I was looking at the images you have in the video and thinking that you'll need also images with the chessboard rolled and pitched, as in an intrinsic calibration.

miguelriemoliveira, 2019-12-07:
This is done, right?

eupedrosa, 2019-12-07:
Yes. This can be closed.

closed 2019-12-07 by eupedrosa