lardemua/atom #44: Tracking of chessboard in 3D

miguelriemoliveira, 2019-11-05:
To use the asus xtion check https://wiki.ros.org/openni2_launch

miguelriemoliveira, 2019-11-12:
First step is done. @tiagomfmadeira can already cluster the point clouds ...

tiagomfmadeira, 2019-11-12:
Here is the result using Open3D's _cluster_dbscan_

[image]

tiagomfmadeira, 2019-11-18:
I have integrated the previously described method of clustering with the rest of the implementation. 
It is now possible to read the clouds from the **/camera/depth/points** topic, extract the clusters, and publish a colored portion of the cloud in /camera/depth/points/labeled.
However, the clustering seems to be too slow, at least with the parameters I'm currently experimenting with... Will look into the clustering method again.

[image]

miguelriemoliveira, 2019-11-18:
Great news! It looks great. The problem of being to slow, can you quantify, how many hz?

We had the idea of clustering in the image space, we could go into that ...

tiagomfmadeira, 2019-11-28:
Attempting a 2D approach to the clustering for optimization purposes using FloodFill and image contours.
[image]

miguelriemoliveira, 2019-12-11:
Hi @tiagomfmadeira ,

about the part of the calibration config json 

 ```json
"top_center_rgbd_camera_depth_frame": {
           "link": "top_center_rgbd_camera_depth_optical_frame",
           "parent_link": "top_center_rgbd_camera_link",
           "child_link": "top_center_rgbd_camera_depth_frame",
           "topic_name": "/top_center_rgbd_camera/depth/points"
        }
```

I would suggest to put the name **top_center_rgbd_camera_depth_frame** to **top_center_rgbd_camera_depth** since it is the name of the sensor.

miguelriemoliveira, 2019-12-11:
Hi again @tiagomfmadeira ,

reviewing your code ... this 

https://github.com/lardemua/AtlasCarCalibration/blob/de74c5f561d7d3cfb68a93008437efd6acc2b76a/interactive_calibration/src/interactive_calibration/interactive_data_labeler.py#L344

looks strange since a depth image should be floar or 16C1, but not 8C1

miguelriemoliveira, 2019-12-16:
http://docs.ros.org/kinetic/api/image_geometry/html/python/index.html

miguelriemoliveira, 2019-12-16:
rospy get param https://wiki.ros.org/rospy_tutorials/Tutorials/Parameters#Getting.2C_Setting.2C_and_Deleting_Parameters

miguelriemoliveira, 2020-01-16:
Complete. Will continue to work on #117.

closed 2020-05-28 by miguelriemoliveira