lardemua/atom #45: Run an optimization with only the  cameras to make sure we did nothing wrong in last Tuesday's changes.

miguelriemoliveira, 2019-11-08:
@afonsocastro , can you do this?

afonsocastro, 2019-11-08:
yes, i can

afonsocastro, 2019-11-08:
yes, i can

afonsocastro, 2019-11-09:
It is all working just fine. 
Here are the results obtained for the optimization with only cameras vs calib camera:
[image]

AVERAGE ERROR (our optimization):
x = 2.20814818259 pix ;   y = 0.732894332321 pix

AVERAGE ERROR (openCV stereo calibration):
x = 285.961612654 pix ;   y = 15.976624504 pix

AVERAGE ERROR (openCV calibrate camera):
x = 6.93868496473 pix ;   y = 6.00126505319 pix

I noticed that the first guess xacro file must already exist since the program will only rewrite it. Because of this, the first calibration ever will not work if the user does not create the xacro before.

README updated with the new command lines.

miguelriemoliveira, 2019-11-11:
The values for the stereo calibration do not make sense ... do you know what is the problem?

afonsocastro, 2019-11-12:
After finding and solving the problem of the stereo calibration implementation, here are the obtained results:

[image]

AVERAGE ERROR (our optimization):
x = 2.20814818259 pix ;   y = 0.732894332321 pix

AVERAGE ERROR (openCV stereo calibration):
x = 1.87823367918 pix ;   y = 0.841562853078 pix

AVERAGE ERROR (openCV calibrate camera):
x = 2.60907725178 pix ;   y = 3.77128716793 pix

They are closest to what it is expected, but are these values good enough?

eupedrosa, 2019-11-12:
What was the error? You should also provide the error deviation to better judge the data. Just the mean error is not enough.

miguelriemoliveira, 2019-11-12:
Hi, yes std would be interesting, but we also get a feeling of the data deviation from the graphic. 
Nonetheless, If you can add the info that would be another plus ...

miguelriemoliveira, 2019-11-12:
Also, I would close this issue since the title seems to have been completely fulfilled...

afonsocastro, 2019-11-16:
std info is already added.
I will commit and close this issue soon

closed 2019-11-16 by afonsocastro