lardemua/atom #48: How to configure catkin build

miguelriemoliveira, 2019-11-12:
Hi @eupedrosa ,

can you tell me what must be done to configure catkin build?

Thanks,
Miguel

miguelriemoliveira, 2019-11-12:
I remember you configured something with /opt/ros ...

eupedrosa, 2019-11-12:
I will leave here the full steps:

### Install
```$ sudo apt install python-catkin-tools```
### First use
In the topmost directory of your code, i.e. the one that has a `src` directory, execute
```
$ catkin configure --extend /opt/ros/melodic
$ catkin build
$ source devel/setup.bash
```
### Onward 
```
$ source devel/setup.bash
```

miguelriemoliveira, 2019-11-12:
BTW, this 

> catkin configure --extend /opt/ros/melodic

should be:

> catkin config --extend /opt/ros/melodic

closed 2019-11-12 by miguelriemoliveira

eupedrosa, 2019-11-12:
Sorry, I wrote it from memory!