lardemua/atom #50: Better 3D plotting?

miguelriemoliveira, 2019-11-13:
Hi @afonsocastro , @tiagomfmadeira and @eupedrosa ,

we are using matplotlib for drawing the 3D graphs, but the quality of the figures is not very good.
Here's an example:

[image]

I looked around and found [this](https://docs.enthought.com/mayavi/mayavi/auto/examples.html)

do you know any good alternatives for 3D plotting with python?

miguelriemoliveira, 2019-11-13:
here's another interesting one
https://plot.ly/python/#3d-charts

eupedrosa, 2019-11-13:
[Plotly](https://plot.ly/)  looks interesting :+1:
Do you want to test it, @miguelriemoliveira ?

miguelriemoliveira, 2019-11-16:
I am on it ...

miguelriemoliveira, 2019-11-16:
So I tried to dive into plotly but could not as of now find a way to update a graph as we need in our application. I posted a question in the forum, lets see...

miguelriemoliveira, 2019-11-21:
After discussion with @eupedrosa and @afonsocastro we decided the best and fastest way to go is to use rviz. Will work on it.

miguelriemoliveira, 2019-11-24:
Made some progress. Right now I am showing the car and sensor models, the tfs, the images and the laser data. Here's an example:

https://youtu.be/M22ZAQuAusw

Will include the chessboards and other inportant features in the near future.

miguelriemoliveira, 2019-11-25:
Now we are drawing the chessboards for each collection ...

[image]

eupedrosa, 2019-11-25:
:+1: nice

commit f6293288 (2019-11-26): Close #50. Drawing almost everything we had before in matplotlib.
closed 2019-11-26 by miguelriemoliveira