lardemua/atom #52: Saving first guess does not work.

eupedrosa, 2019-11-15:
Now that we are not using URDF parser (with custom entities), saving the URDF with the first guess does not work. We have to revise this. 

Does it make sense to save the first guess to a urdf file? Or maybe we should add the first guess to the json configuration file?

I understand that the initial idea was to augment the urdf but using the json would be simpler (in my opinion). However, I still believe that the final output could be the xacro/urdf.

miguelriemoliveira, 2019-11-15:
Hi @eupedrosa ,

we can discuss on Tuesday if it makes sense. 

@afonsocastro what do you think?

afonsocastro, 2019-11-16:
I think the approach should be as clean, organized and homogeneous as possible. This means that everything should make sense. 
I agree with the idea of creating a JSON file to store the poses first guess. The philosophy is: files that are created/changed by the calibration procedure are always JSON, and robot xacro files are "untouchable". 
The goal is to not change the robot description structure, so if the final output could actually be the urdf xacro, then yeah I follow the JSON idea. We have to test it to see if it is possible.

miguelriemoliveira, 2019-11-18:
Hum, I am not so keen as you in going into the json. I am not sure it will all work without us having to do a specialized json parser for this ... In any case, we all agree this is a low priority one right?

eupedrosa, 2019-11-18:
Tomorrow (19 nov 19), I will try to sell you, @miguelriemoliveira, the json approach. 

I've realized that outputting an `.urdf` as the final result of the calibration is something that I would like to have. Just like you said it would be. However, I think that the intermediate steps of the calibration should stay out of the xacro/urdf. It seems that @afonsocastro also agrees.

>  In any case, we all agree this is a low priority one right?

For my use case this is actually middle to high priority. Because I do not know were the camera is relative to the base of the arm I have to guess where it is. The first guess is very useful for this. But it has some problems. It is difficult to set a first guess for a camera in 3D space, being the distance from the sensor to the chessboard the more difficult. I struggled a little with that. 

Because of this "troubles" I have a few ideas to improve the first guess. One of them is to use the chessboard and `solvePnP` to help with the first guess. We can discuss more tomorrow.

miguelriemoliveira, 2019-11-21:
@eupedrosa says problem solved with #53

closed 2019-11-21 by miguelriemoliveira