lardemua/atom #55: Reorganize optimization dictionaries

miguelriemoliveira, 2019-11-25:
Hi @afonsocastro 

I am doing the visualization and getting really confused about the dictionaries we have in our optimization.

So there are:

1. dataset_sensors, which is read from the file, contains keys **sensors** and **collections**
(It is weird that a **dataset_sensors** contains a key **sensors**)
2. We then augment the dataset_sensors by adding another key, **chessboards**
3. Then we have a "**dataset_chessboard_points**" ... why a different dataset?
4. We also have a **dataset_graphics** (this makes sense, it is completly independant)

I wanted to change but I think it may have impact not only on our optimization (which I can fix) but in your evaluation code ...

What do you think?

eupedrosa, 2019-11-28:
*Dictionary* Everything!!! :smile:

miguelriemoliveira, 2019-11-29:
Actually I am all for it. We don't have to try to memorize complex data structures, they're all dictionaries.

And you can just save them to a json on the fly ...

miguelriemoliveira, 2020-06-14:
Done.

closed 2020-06-14 by miguelriemoliveira