lardemua/atom #56: matplotlib in the optimizer code base.

eupedrosa, 2019-11-25:
I noticed that the `Optimizer` class depends on `matplotlib`.
What is the reasoning behind this coupling? 
For example, in #50 RViz is being used for visualization, however a `matplotlib` figure stills *pollutes* the screen. Should we try to decouple this?

miguelriemoliveira, 2019-11-25:
This is for displaying the figure showing the residuals, which is a figure detached from the particular optimization problem you have configured. 

So I think at least for now this should stay as is...

miguelriemoliveira, 2019-11-25:
If you agree please close ...

eupedrosa, 2019-11-25:
Ok, I understand. For now it is useful to stay as it is. In the future we may address this issue again if necessary.

closed 2019-11-25 by eupedrosa