lardemua/atom #57: Location, location, location...

eupedrosa, 2019-11-25:
In which package should I put an the *arm_and_camera* optimization problem? In the `OptimizationUtils`  or  `interactive_calibration`?
Being a calibration problem, I think that is should go to the `interactive_calibration` package. What is your opinion?

miguelriemoliveira, 2019-11-25:
Yes, you are doing a code which is parallel to the
https://github.com/miguelriemoliveira/OptimizationUtils/tree/master/test/sensor_pose_json_v2
folder

Regards,
Miguel

*  Miguel Riem de Oliveira                                             *
email:    mriem@ua.pt    |    m.riem.oliveira@gmail.com
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2019-11-26:
We talked about this today.

closed 2019-11-26 by miguelriemoliveira