lardemua/atom #64: Data collector should  consider min/max laser ranges

eupedrosa, 2019-11-26:


miguelriemoliveira, 2019-11-28:
For reference, while this is not working we have to remove from the calibration_26nov2019 dataset collections:

5 6 7 12 13 14 15

miguelriemoliveira, 2019-11-28:
Also consider the problem of removing a percentage of points at the border.

```
threshold_to_remove = 0.03  # remove 10% of data from each side
            number_of_idxs = len(clusters[idx_closest_cluster].idxs)
            idxs_to_remove = int(threshold_to_remove * float(number_of_idxs))
            clusters[idx_closest_cluster].idxs_filtered = clusters[idx_closest_cluster].idxs[
                                                          idxs_to_remove:number_of_idxs - idxs_to_remove]

            self.labels['idxs'] = clusters[idx_closest_cluster].idxs_filtered
```

commit 6264bec3 (2019-11-30): Close #64. Retooled the LaserScan clustering mechanism to try to solve these problems. We are now not removing any % of points from the left and right side of he cluster, with the caveat that sometimes a points not lying entirely on the calibration pattern's plane is included.
closed 2019-11-30 by miguelriemoliveira