lardemua/atom #65: New intrinsic calibration of both cameras

miguelriemoliveira, 2019-11-26:
Hi @afonsocastro ,

this is important to make sure we do not have problems from this part. The ideal case would be to calibrate the intrinsics using the bagfile ... can you try?

afonsocastro, 2019-11-26:
Yes, I will try with the bagfile

miguelriemoliveira, 2019-11-26:
Great!

afonsocastro, 2019-11-27:
Bad news: our bagfile isn't good enough to enable an intrinsic calibration...  
After all bag recorded time, and 33 samples added by the ROS camera calibrator, here's the calibrator:

[image]

X and Y coordinates and the chessboard size haven't changed sufficiently. This occurred because this bag was recorded with the lasers continuous chessboard detection concern. That's why the Y coordinate of the chessboard origin didn't suffer big changes.

So, we must calibrate the cameras live, in the real scene... Or at least record a bagfile with the specific purpose of getting an intrinsic calibration

miguelriemoliveira, 2019-11-28:
You could just use a matlab camera calibration which reads from a folder of png images

https://www.mathworks.com/videos/camera-calibration-with-matlab-81233.html

and then adapt the output of matlab to produce the yaml.

But I think it is best to calibrate the camera using ros.

afonsocastro, 2019-11-28:
Ok, I just uploaded the yaml files with the values of the matlab intrinsic calibration of each camera.
They have less information than the yaml files returned by the ROS calibration, but I think we only need the intrinsic matrix and the distortion parameters. 
I am not very confident about this, we can discuss it tomorrow

miguelriemoliveira, 2019-11-28:
Calibrate in ROS 3 times in a row to make sure we have a good calibration.

afonsocastro, 2019-12-03:
First attempt:

```yaml
image_width: 1624
image_height: 1224
camera_name: 14233704
camera_matrix:
  rows: 3
  cols: 3
  data: [1420.060841543088, 0, 794.7046593573046, 0, 1421.214911871335, 564.6352822046543, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.2086707440041696, 0.1741483019004053, -0.002057879830518041, -0.0002302521057234884, 0]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [1344.62646484375, 0, 793.1936514067638, 0, 0, 1370.65185546875, 559.9345065366651, 0, 0, 0, 1, 0]
```

afonsocastro, 2019-12-03:
http://wiki.ros.org/openni_launch/Tutorials/IntrinsicCalibration

afonsocastro, 2019-12-04:
TOP LEFT CAMERA:

1 attempt:
```yaml
image_width: 964
image_height: 724
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [1250.153116, 0.000000, 439.243955, 0.000000, 1263.561324, 369.208600, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.118686, 0.142313, 0.001759, -0.008291, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [1228.778564, 0.000000, 433.386312, 0.000000, 0.000000, 1253.042114, 369.897945, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```
2 attempt:

```yaml
image_width: 964
image_height: 724
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [1137.131374, 0.000000, 440.105345, 0.000000, 1150.798890, 352.399934, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.167092, 0.089636, 0.003106, -0.003480, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [1102.939331, 0.000000, 435.452355, 0.000000, 0.000000, 1132.079346, 353.244830, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

3 attempt:

```yaml
image_width: 964
image_height: 724
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [1214.399004, 0.000000, 460.072829, 0.000000, 1228.512287, 376.838308, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.151545, 0.222829, 0.001664, -0.005372, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [1190.538940, 0.000000, 456.199821, 0.000000, 0.000000, 1214.088379, 377.703372, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

afonsocastro, 2019-12-04:
TOP RIGHT CAMERA

1 attempt:
```yaml
image_width: 964
image_height: 724
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [1127.135827, 0.000000, 507.760303, 0.000000, 1146.647310, 356.223224, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.189441, 0.248208, -0.000588, -0.001458, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [1096.488525, 0.000000, 507.761502, 0.000000, 0.000000, 1127.231934, 355.835772, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```
2 attempt:

```yaml
image_width: 964
image_height: 724
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [1170.443468, 0.000000, 481.006460, 0.000000, 1182.755708, 372.142310, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.212561, 0.324017, 0.001130, 0.000481, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [1137.985352, 0.000000, 481.284349, 0.000000, 0.000000, 1161.854126, 372.851654, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

3 attempt:

```yaml
image_width: 964
image_height: 724
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [1158.660264, 0.000000, 486.058889, 0.000000, 1170.428824, 389.607847, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.205536, 0.489565, 0.003190, 0.000423, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [1137.452026, 0.000000, 486.280956, 0.000000, 0.000000, 1151.576538, 391.340611, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

afonsocastro, 2019-12-04:
TOP CENTER RGB-D CAMERA (RGB)
1 attempt:
```yaml
image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [552.683663, 0.000000, 330.408026, 0.000000, 560.974606, 238.468676, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.117068, -0.012465, 0.003419, 0.008242, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [582.377136, 0.000000, 334.351958, 0.000000, 0.000000, 593.400391, 239.906599, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

2 attempt:
```yaml
image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [578.360287, 0.000000, 343.154551, 0.000000, 586.694903, 252.450873, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.189760, -0.301085, 0.014456, 0.014631, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [597.512024, 0.000000, 351.996258, 0.000000, 0.000000, 610.772888, 258.252418, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

3 attempt:
```yaml
image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [574.885960, 0.000000, 323.315427, 0.000000, 582.044634, 242.217345, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.081895, 0.043307, 0.005784, 0.004452, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [602.349976, 0.000000, 325.475800, 0.000000, 0.000000, 609.730713, 244.493108, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

afonsocastro, 2019-12-04:
About this last 3 intrinsic calibrations:
1 - All four calibrator parameters (X, Y, Size, Skew) were green = enough different detected chessboard poses
2 - all intrinsic calibrations (3 per rgb camera) were computed with a database of (more or less) 50 image samples
3 - any of the chess corners was detected misplaced: whenever that happened, I aborted the calibration and started a new one (reference to the discussion in #70  )

afonsocastro, 2019-12-04:
It wasn't yet possible to perform the intrinsic calibration of the depth camera (top_center_rgbd_camera). The related problems are discussed in #78

miguelriemoliveira, 2019-12-04:
Hi @afonsocastro ,

very nice work, and great work also on documenting what you did.

Analysing the results I noticed we get very different values in the 3 calibrations.
For example, for **top_left_camera**, **fx** (zoom factor along x axiss, which is one of the parameters that most influences the projections) is **[1250.153116 ; 1137.131374  ; 1214.399004]**

Average is 1200 and **standard deviation is 57** which is a lot I think (but not sure).

@eupedrosa do you have this feeling concerning intrinsic calibrations? I could ask some of my senior colleagues if this is normal. 

Which values should we use?

closed 2019-12-05 by afonsocastro