lardemua/atom #66: Python style guide.

eupedrosa, 2019-11-27:
Yesterday, @miguelriemoliveira, mentioned that he is using [PEP8](https://www.python.org/dev/peps/pep-0008/) style guide for python code. In my opinion we should all adopt this style guide. 

What do you all think? Or maybe I was the only one not following pep8.

miguelriemoliveira, 2019-11-27:
Hi,

To the best of my knowledge, Tiago and Afonso are also using pycharm and
thus should be compliant with pep8.

There is only one thing I can recall where we are not complying: function
names should be lowercase.

We may change this in the future.

eupedrosa, 2019-11-27:
As I suspected. It was only me.
I will close this issue and assume that pep8 should be respected.

closed 2019-11-27 by eupedrosa