lardemua/atom #69: [OptimizationUtils] Optimizer will fail if visualization function is not set.

eupedrosa, 2019-11-29:
I got the following error:

```
Traceback (most recent call last):
  File "/home/efp/+play/lardem-ua/src/OptimizationUtils/test/eye_in_world_calib.py", line 151, in <module>
    main()
  File "/home/efp/+play/lardem-ua/src/OptimizationUtils/test/eye_in_world_calib.py", line 143, in main
    opt.callObjectiveFunction()
  File "/home/efp/+play/lardem-ua/src/OptimizationUtils/./OptimizationUtils/OptimizationUtils.py", line 246, in callObjectiveFunction
    return self.internalObjectiveFunction(self.x)
  File "/home/efp/+play/lardem-ua/src/OptimizationUtils/./OptimizationUtils/OptimizationUtils.py", line 260, in internalObjectiveFunction
    if self.always_visualize and self.vis_counter >= self.vis_niterations:
AttributeError: Optimizer instance has no attribute 'always_visualize'
```

Is forcing a visualization function an intended behavior?

miguelriemoliveira, 2019-11-29:
No, not really. How are you launching you code?

Are you using the command line parameter -vo (--view_optimization)?

eupedrosa, 2019-11-29:
I am trying to create a new optimization script for the *eye in the world* problem. Not the ones that already exist. I already understand what is happening, not a big issue. The variable `self.always_visualize` is not initialized at the `__init__` function.

miguelriemoliveira, 2019-11-29:
OK, but if it was me I would just copy paste to maintain a similar structure and then change what was needed.

miguelriemoliveira, 2019-11-29:
Great, please close if solved.

eupedrosa, 2019-11-29:
True, but that way I will not learn. You already know how it works. Anyhow, I am trying to follow the structure that you already have. The filtering stuff is very useful.

closed 2019-11-29 by eupedrosa

miguelriemoliveira, 2019-11-29:
Great! My goal was just to have the more silimar structure as possible because, remember, in the future, the objective could be to merge both these codes to do sensor extrinsic + hand in eye + eye in world calibration all in one.