lardemua/atom #70: Chessboard detection is not invariant to orientation.

eupedrosa, 2019-11-29:
I noticed that, while collecting data, the chessboard detection can have different orientations. Look at the bottom right of the images:
[image]

[image]

Is this something that you are aware of?

miguelriemoliveira, 2019-11-29:
I was aware of this and this is a problem: if it happens on the dataset it will ruin your calibration (intrisinc, extrinsic, whatever).

We have a calibration pattern with oddxeven number of corners. This I think makes the detection more robust to these situations. 

However, whenever we note this occurs, we just thrash the dataset and start a new one.

@afonsocastro  you should also be carefull with this when doing #68

miguelriemoliveira, 2019-11-29:
@eupedrosa is your chessboard odd x even (or  even x odd) or is it even x even (or odd x odd)?

eupedrosa, 2019-11-29:
It is a odd x odd chessboard. Do I have to change it? I think I can try to solve this with software!!!!

miguelriemoliveira, 2019-11-29:
Changing it will improve I think  but I am not sure it will solve the whole problem.

Solving with software is cumbersome since you would have to make some kind of assumption which is not general and, furthermore, you would have to add this new software to the opencv chessboard detection code, and ask for a pull request.

If you say solve with software only for our calibration (which means you compute the intrinsics with another dataset where this phenomenon does not occur), we could do that ...

How often do you have this issue? I mean, we just do not collect data when the chessboards are not well detected ...

eupedrosa, 2019-11-29:
It happens way to often.
I solved it with a *hack*. It is not very ugly. It adds 3 lines of code and 1 line change. However, it adds the assumptions that the chessboard will maintain an angle bellow 90 degrees. Is this bad?

miguelriemoliveira, 2019-11-29:
Where is this hack located?

miguelriemoliveira, 2019-11-29:
I like the "not very ugly" expression ...

eupedrosa, 2019-11-29:
The hack is located at `interactive_data_label.py`.
Check line 232
```python
           # Find chessboard corners
            self.found, corners = cv2.findChessboardCorners(image_gray, (self.numx, self.numy))
            if not self.found:
                cv2.drawChessboardCorners(image, (self.numx, self.numy), corners, self.found)  # Draw and display the corners

            if self.found is True:
                diff = corners[0][0][0] - corners[self.numx-1][0][0]
                if diff > 0:
                    corners = corners[::-1]

                cv2.drawChessboardCorners(image, (self.numx, self.numy), corners, self.found)  # Draw and display the corners

                # print('Found chessboard for ' + self.name)
                criteria = (cv2.TERM_CRITERIA_EPS + cv2.TERM_CRITERIA_MAX_ITER, 30, 0.001)
                corners2 = cv2.cornerSubPix(image_gray, np.array(corners), (self.numx, self.numy), (-1, -1), criteria)
                corners2_d = []
```
It just verifies the x coordinate difference between the first corner and the last corner in the same row.
If the difference is positive it means that it is flipped. I just reverse the corner array.

miguelriemoliveira, 2019-11-29:
It looks great. Does it work? Is the drawing rotated back to the propper place? Do you think it may have problems in other scenarios?

If it works my opinion is to include it. I would just recomend you do a ros warn every time you do this, to notify the user we are doing it.

eupedrosa, 2019-11-29:
Yes, it draws the chessboard in the proper place. I do not thing it may cause any problem beyond the one that I already mention. The ROS warn is a good idea. I will test it a little bit more.

miguelriemoliveira, 2019-11-29:
Sounds great. Can you post some images showing you solved the problem of
the images you posted prevuously?

eupedrosa, 2019-11-29:
I give you moving images (without the hack and with the hack)
[image] [image]

Not synchronized but..

miguelriemoliveira, 2019-11-29:
This is really nice, gifs. I got to do that as well. 

Not syncronized ... booo !

miguelriemoliveira, 2019-11-29:
Reopenning, the commit should close the issue ...

miguelriemoliveira, 2019-11-30:
Hi @eupedrosa ,

Just remembered we also will have this problem in 

https://github.com/miguelriemoliveira/OptimizationUtils/blob/20058582f784469088dd621f61e55c97d6307ef8/test/sensor_pose_json_v2/chessboard.py#L227

Actually, the idea already in a json there was not to redetect the chessboard but to use the corners from the initial detection contained in the json.

@eupedrosa can you do this with @afonsocastro 's help?

miguelriemoliveira, 2019-12-03:
@eupedrosa  and @afonsocastro , just did it what I suggested above. Closing issue.

commit 87b8af54 (2019-12-03): Close #70. Using chessboard detection from the labelling stage, rather than redetecting.
closed 2019-12-03 by miguelriemoliveira