lardemua/atom #72: Data collector not saving the data that is labelled

miguelriemoliveira, 2019-12-01:
This is a big bug. When looking at the 

https://github.com/lardemua/AtlasCarCalibration/blob/6264bec39e22645006f33b4346872adada4a1ef4/interactive_calibration/src/interactive_calibration/data_collector_and_labeler.py#L137

in particular 

https://github.com/lardemua/AtlasCarCalibration/blob/6264bec39e22645006f33b4346872adada4a1ef4/interactive_calibration/src/interactive_calibration/data_collector_and_labeler.py#L151

I noticed that we were actually saving sensor data received in messages that were not labelled, since the labelling occurs here

https://github.com/lardemua/AtlasCarCalibration/blob/6264bec39e22645006f33b4346872adada4a1ef4/interactive_calibration/src/interactive_calibration/interactive_data_labeler.py#L112

Will try to fix this ...

miguelriemoliveira, 2019-12-01:
Hi @eupedrosa ,

the problem here is that you have the timercallback inside the **interactive_data_labeller** periodically clustering the data, but when the user saves a collection the **data_collector_and_labeller** actually waits for new messages which have not been labelled. 

Perhaps it is not noticeable for high streaming frequencies, but we should save to the json the sensor data message with the corresponding labels for that particular message, not for a previous one.

Is it clear now?

eupedrosa, 2019-12-01:
Now I understand.
This explains something that happened to me. I clicked to collect data when the chessboard was properly detected but the one that was actually save failed to detect. I was using a replay with a rate of 0.5.

miguelriemoliveira, 2019-12-01:
Solved. @eupedrosa if you see again what you described let me know ... it should not happen ...

commit 786b2a42 (2019-12-01): Fixes #72 and Fixes #73.
closed 2019-12-01 by miguelriemoliveira