lardemua/atom #73: Not saving up to date transforms in json file!!!

miguelriemoliveira, 2019-12-01:
Hi @afonsocastro and @eupedrosa ,

I was looking at the https://github.com/lardemua/AtlasCarCalibration/blob/master/interactive_calibration/src/interactive_calibration/data_collector_and_labeler.py

and the 
https://github.com/lardemua/AtlasCarCalibration/blob/master/interactive_calibration/src/interactive_calibration/interactive_data_labeler.py

and noticed a problem of that the data being labelled was not hte one being saved. Check #72 

But then I explored a bit more and noticed that the transforms being saved to the collections are also not being updated. They are only updated in the data_collector_and_labeller.py's constructor and then never again ...

https://github.com/lardemua/AtlasCarCalibration/blob/6264bec39e22645006f33b4346872adada4a1ef4/interactive_calibration/src/interactive_calibration/data_collector_and_labeler.py#L125

That means we were using the transform we collected from the start, never updating them again ... how did this work? @afonsocastro , I think it was because all our transform were static, right?

In @eupedrosa 's case (#42 ) this would not work ... 

I am not sure I am well today, I see bugs everywhere : - )

eupedrosa, 2019-12-01:
For the *eye in world* it is a problem if the transformations are not correct. I did not notice because I was not there yet. In fact, it was my next step.

I am yet to understand the problem in #72. Last week we did some data collection, right? And it was working, or not?

miguelriemoliveira, 2019-12-01:
@tiagomfmadeira you should also know about this since you are using the collector for 3d points clouds. Also, check #72 ...

miguelriemoliveira, 2019-12-01:
> For the _eye in world_ it is a problem if the transformations are not correct. I did not notice because I was not there yet. In fact, it was my next step.
> 
> I am yet to understand the problem in #72. Last week we did some data collection, right? And it was working, or not?

It worked (more or less) because the sensor frequencies are high and we only store collections after we see a static scene for a couple of seconds. Then again, perhaps the errors we've seen (or part of then) were related to this ...

eupedrosa, 2019-12-01:
Well, this needs to be fixed. The transformations are the same for all collections. This is OK for AtlasCar but not OK for the arm, for obvious reasons.

I believe that may be another problem. Time synchronization. The transformations should be from the same (or approximate) timestamp as the rest of the data. This may not be necessary if we assume that the user collects data when the system is steady. Maybe we can go with the latter approach and solve this latter on.

miguelriemoliveira, 2019-12-01:
Yes, for now we just collect tfs all at the same time. I am on it ...

Latter on we may try to devise a less naive approach, but the idea was to assume that the environment was static arround the moment of collecting the data.

It should be enough I think ...

miguelriemoliveira, 2019-12-01:
Solved, @eupedrosa when you can please try it out with the UR10.

eupedrosa, 2019-12-02:
The transformations are empty in JSON file. I guess that @miguelriemoliveira's common cold was underestimated.

closed 2019-12-02 by eupedrosa

miguelriemoliveira, 2019-12-02:
Thanks. I did not understand you had solved it so I went to look at it and changed a bit.
The main thing was as you say to read the abstract transforms after building the list of sensors...

commit 786b2a42 (2019-12-01): Fixes #72 and Fixes #73.
commit 87b37420 (2019-12-02): Fix #73. The transforms are only collected after the sensors are defined.
commit 5e2c5b30 (2019-12-02): Fixed #73. Should be working now ...