lardemua/atom #75: Compressed Depth image and republish

afonsocastro, 2019-12-03:


miguelriemoliveira, 2019-12-03:
From what I read here this should work. I will try once I have time and a camera nearby.

https://wiki.ros.org/compressed_depth_image_transport

eupedrosa, 2019-12-03:
You should also take a look at this https://wiki.ros.org/dynamic_reconfigure
Specially the `dynparam` command-line tool. It can be uses to change the parameters.

miguelriemoliveira, 2019-12-03:
Thanks, but this is the same functionality as when we did:

`rosparam set/get <param_name> <param_value>`

https://wiki.ros.org/rosparam

Or am I missing something?

miguelriemoliveira, 2019-12-03:
@afonsocastro since you are at the atlascar can I also ask you to check the frequency of the topic

/top_center_rgbd_camera/depth/image/compressedDepth (or something similar)

using a 

rostopic hz <topic name>

afonsocastro, 2019-12-04:
frequency = 3.6 hz ?

[image]

miguelriemoliveira, 2019-12-04:
Yes, this is the problem we had yesterday, the frequency is too slow ...

eupedrosa, 2019-12-04:
>Thanks, but this is the same functionality as when we did:
>
>rosparam set/get <param_name> <param_value>
>
>https://wiki.ros.org/rosparam
>
>Or am I missing something?

You are missing something, `dynparam` is used to tell the a node to load the parameters from the param server.

https://wiki.ros.org/dynamic_reconfigure

miguelriemoliveira, 2019-12-05:
Closed by https://github.com/lardemua/atlas-core/commit/417ca9e5aef489ca12851b61d3bf7f607b3c3772

closed 2019-12-05 by miguelriemoliveira