lardemua/atom #76: Change the name of the frontal_rgbd_camera to top_frontal_rgbd_camera

miguelriemoliveira, 2019-12-03:
Hi @afonsocastro ,

a refactor should help to do this. I can help.

miguelriemoliveira, 2019-12-03:
I am on it ...

miguelriemoliveira, 2019-12-03:
Done, changed to "top_center_rgbd_camera" to be more consistent with "top_left_camera". Cannot verify if it is working, @afonsocastro , can you test tomorrow?

If it works please close the issue ...

afonsocastro, 2019-12-03:
I will test it now, im working on atlascar

afonsocastro, 2019-12-03:
It all works! but rviz gives a warning:

[image]

miguelriemoliveira, 2019-12-03:
I think that was there before this afternoon (@eupedrosa can you conform?), so I would say the refactor worked.

Waiting for @eupedrosa to close. 

Thanks @afonsocastro !

eupedrosa, 2019-12-04:
It is a strange warning. I think the warning we saw was different. Maybe it is worth trying to understand what is happening.

closed 2019-12-05 by afonsocastro