lardemua/atom #78: Depth camera cannot detected chessboard

afonsocastro, 2019-12-04:


afonsocastro, 2019-12-04:
Tried to calibrate according to http://wiki.ros.org/openni_launch/Tutorials/IntrinsicCalibration . 
Here is the calibrator:

[image]

It never detected the chessboard.
I tried without any light source, but also nothing... Tried with only the car presence lights (cover with paper, to diffuse the speckles), but still didn't detected the chessboard.

After that, and following the calibration tutorial, i only launched the rgb-d camera driver. In that way, there weren't any other nodes subscribing to RGB images besides the 3d camera. It also didn't solved the problem

miguelriemoliveira, 2019-12-04:
Hi @afonsocastro , which command do you use?

Also, did you cover the IR projector as stated here "The simplest solution is to cover the projector (lone opening on the far left) with one or two Post-it notes, mostly diffusing the speckles"

eupedrosa, 2019-12-04:
@afonsocastro, I think that you are using the wrong image. You should use the `/camera_name/ir/image` topic (or something similar).

miguelriemoliveira, 2020-07-18:
Too old. Closing.

closed 2020-07-18 by miguelriemoliveira