lardemua/atom #79: Generate point clouds when playing back bagfile

miguelriemoliveira, 2019-12-06:


miguelriemoliveira, 2019-12-06:
Hi @eupedrosa ,

I think we should have saved the 

` /top_center_rgbd_camera/depth/image_raw/compressedDepth`

instead of

` /top_center_rgbd_camera/depth/image/compressedDepth`

Now I am trying to generate the point clouds using the openni2_launch with the "load_drivers" argument set to false, and the output is:

> [ WARN] [1575628804.597025590, 1575574008.526946402]: [image_transport] Topics '/top_center_rgbd_camera/depth/image_raw' and '/top_center_rgbd_camera/depth/camera_info' do not appear to be synchronized. In the last 10s:
> 	Image messages received:      0
> 	CameraInfo messages received: 150
> 	Synchronized pairs:           0

So they are waiting for the image_raw rather than the image ,,,

will keep looking into it.

eupedrosa, 2019-12-06:
You should not use `openni2_launch`. I would recommend to use the node `depth_image_proc/point_cloud_xyz` from http://wiki.ros.org/depth_image_proc.
This way we can skip the openni2 in the replay.

miguelriemoliveira, 2019-12-06:
Yep, I was investigating just that, however there are nodelets rather than nodes and it becomes a whole different thing. 

[image]

I would have to launch a nodelet manager, then plugin the propper nodelet, I have done this kind of stuff in the RACE project and it is quite dificult.

I would vote for taking another bagfile with the depth/image_raw and then just use all the setup that we already have set up in openni2_launch, it is much easier ...

miguelriemoliveira, 2019-12-06:
So, if I would be trying to get a new bag file next week (I will try to come help Afonso), perhaps I will also try to improve the chessboard with the velcro or velvet idea...

eupedrosa, 2019-12-06:
It's not that hard..
By the way, did you try remapping the topic to `../image_raw` ? Try using `image_raw` as the topic output in the `republish` node.

miguelriemoliveira, 2019-12-06:
Just setup a nodelet manager and a nodelet to generate point clouds, making this read from the depth/image (rather than the usual depth/image_raw) , it launches but when I subscribe to the point cloud the nodelt aborts with the message 

> [ERROR] [1575630722.500663717, 1575573833.860206039]: Client [/standalone_nodelet] wants topic /top_center_rgbd_camera/depth/image to have datatype/md5sum 
> [sensor_msgs/CameraInfo/c9a58c1b0b154e0e6da7578cb991d214], but our version has > 
> [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743]. Dropping connection.

I am guessing he's complaining he's complaining that the depth image is float32 rather that the required uint16, which, in other words is, he's waiting for depth/image_raw rather than /depth/image

eupedrosa, 2019-12-06:
It is complaining about message type mismatch. It was expecting a `CameraInfo` and it got an `Image`.

miguelriemoliveira, 2019-12-06:
> It's not that hard..

done.

> By the way, did you try remapping the topic to `../image_raw` ? Try using `image_raw` as the topic output in the `republish` node.

did it now and it works!

[image]

Colored point cloud does not work, but I think this is more than enough for @tiagomfmadeira 

In any case, @eupedrosa I think we should fix it to save in the bag the image_raw data. You agree?

@afonsocastro , my fault (read above) but we are probably going to need another bag file ... can we do it toghether next week? Can we try to use the velcro / velvet approach to cover the black squares? 

Check this image, often you don't have almost any of the black squares hidden from the laser data ...

[image]

@tiagomfmadeira you may continue by subscribing to the topic **/top_center_rgbd_camera/depth/points**

miguelriemoliveira, 2019-12-06:
Thanks for the quick help @eupedrosa

eupedrosa, 2019-12-06:
> In any case, @eupedrosa I think we should fix it to save in the bag the image_raw data. You agree?

I agree.

> Thanks for the quick help @eupedrosa

:+1:

miguelriemoliveira, 2019-12-06:
https://github.com/lardemua/atlas-core/commit/aad11187edcb64ae13238e7f51b2739087906cff

closed 2019-12-07 by miguelriemoliveira