lardemua/atom #80: Houston, we cannot transform to root.

eupedrosa, 2019-12-06:
I need to get the transformation from the `chessboard_link` to the root frame.  I believe there is _no_ function in the Optimization package that provides this functionality. The `getAggregateTransform()` only works if the parent is the root link. This works for AtlasCar, but not for the arm that has several links in the middle.

miguelriemoliveira, 2019-12-06:
Roger that. Only tonight or tomorrow I will be able to take a look. In the
meantime you can meditate in deep space ...

eupedrosa, 2019-12-06:
I have an outer space solution.
Use https://github.com/safijari/tiny_tf to replicate ROS tf. I will work on this :smiley:

miguelriemoliveira, 2019-12-07:
Hi @eupedrosa 

From what you describe

>  need to get the transformation from the chessboard_link to the root frame.
> The getAggregateTransform() only works if the parent is the root link. 

I think it should work. Just provide the propper chain. The function is like this:

```
def getAggregateTransform(chain, transforms, mode=0):
    AT = np.eye(4, dtype=np.float)

    if mode == 0:
        for link in chain:
            parent = link['parent']
            child = link['child']
            trans = transforms[link['key']]['trans']
            quat = transforms[link['key']]['quat']
            parent_T_child = translationQuaternionToTransform(trans, quat)
            # print(parent + '_T_' + child + ' =\n' + str(parent_T_child))
            AT = np.dot(AT, parent_T_child)
```

So I would just get a chain from the chessboard_link to the root frame. The chain is actually a list of dictionaries. Here  is an example of how its built.

https://github.com/lardemua/AtlasCarCalibration/blob/7c7b6958fc8457b8e4d73b8c1e747bc69e2caa0d/interactive_calibration/src/interactive_calibration/data_collector_and_labeler.py#L89-L98

```
            # Get the kinematic chain form world_link to this sensor's parent link
            chain = self.listener.chain(value.link, rospy.Time(), self.world_link, rospy.Time(), self.world_link)

            chain_list = []
            for parent, child in zip(chain[0::], chain[1::]):
                key = self.generateKey(parent, child)
                chain_list.append({'key': key, 'parent': parent, 'child': child})

            sensor_dict['chain'] = chain_list  # Add to sensor dictionary
            self.sensors[sensor_key] = sensor_dict
```

Now if you change **self.world_link** and **value.link** by what you need, and then call getAggreateTransforms it should work.

miguelriemoliveira, 2019-12-07:
Actually, I can do a function **GetAggregateTransformFromFrames** (the only thing I miss in python from c++ is function overload) which will create the chain and call **getAggregateTransforms**.

Should I do it?

miguelriemoliveira, 2019-12-07:
Hi again,

after sleeping on it I realized that you are doing this in your optimizer code and as such you don't have access to the ros ecosystem anymore, so no tfs, so this solution 

https://github.com/lardemua/AtlasCarCalibration/blob/7c7b6958fc8457b8e4d73b8c1e747bc69e2caa0d/interactive_calibration/src/interactive_calibration/data_collector_and_labeler.py#L89-L98

will not work.

What we need is to compute the chain of transforms from any given two reference links and a pool of transforms. 

I will do it using graphs, should be done in 30 mins (this is really interesting and my wife is still asleep :) ).

eupedrosa, 2019-12-07:
We will then compare solutions. As I mentioned before, I used a package I
found in GitHub that replicates the TF outside ROS.
I already have code that is optimizing :)
But I only achieve it moments before turning off the computer, so I have to
make more test to make sure it is working properly.

miguelriemoliveira, 2019-12-07:
Moments before fallin asleep? Should be full of bugs : -)

Yes, great. We will compare then. I am having a lot of fun with this and already have the restored tf tree from the json file (no ros ecosystem):

[image]

miguelriemoliveira, 2019-12-07:
I think its done. Took a while because we had to consider also the possibility of the chain providing the inverse transforms when compared to the existing ones.

Should be working. I tested with:

```
transform_pool = dataset_sensors['collections']['0']['transforms']
    print(utilities.getTransform('top_left_camera', 'top_center_rgbd_camera_depth_optical_frame', transform_pool))
```

and it worked fine. For example when doing 

```
transform_pool = dataset_sensors['collections']['0']['transforms']
    print(utilities.getTransform(''top_center_rgbd_camera_depth_optical_frame', 'top_center_rgbd_camera_depth_optical_frame', transform_pool))
```

I get the expected identity 4x4 matrix .

Tested with some other frames and they work fine.

You must make a new dataset, cannot use previous ones. See #54 

Tell me if it worked.

miguelriemoliveira, 2019-12-19:
Working on #98, closing this one.

closed 2019-12-19 by miguelriemoliveira