lardemua/atom #82: Do you like my Pose?

eupedrosa, 2019-12-07:
Hi,
Good News Everybody !!
I got an initial version of the optimization running (and working!!!). Here is a short clip with the poses provided by the optimization.
[image]

eupedrosa, 2019-12-07:
For reference, this was the *first guess*
[image]

miguelriemoliveira, 2019-12-07:
This is really nice. In the first video, the overlap of the virtual robot arm with the video from the calibrated camera looks fantastic. Congratullations!

One more paper to write :-)?

miguelriemoliveira, 2019-12-07:
For writting a paper it would be nice to get a calibration usin standard tools (visp eye in world). You said your students succsessfully did this?

also, the comparison should be numeric, so you should compute the reprojection error for all collections and plot it. 

@afonsocastro did this before, perhaps he can tell you how to do it  really quick ...

miguelriemoliveira, 2019-12-07:
@tiagomfmadeira you should see this as well ...

miguelriemoliveira, 2019-12-07:
white background everywhere for the papers please :-)

eupedrosa, 2019-12-07:
> This is really nice. In the first video, the overlap of the virtual robot arm with the video from the calibrated camera looks fantastic. 

I was surprised that it work so well. I used 25 collections and it took 16 seconds to optimize.
But there are a few details we have to discuss before moving forward. Next Tuesday would be nice.

> For writting a paper it would be nice to get a calibration usin standard tools (visp eye in world). You said your students succsessfully did this?

I have to re-check this next week. Last time I asked them they said *yes*, but I need more details.

afonsocastro, 2019-12-08:
This looks very nice!! The poses really improved from the first guess. Yes, I can help with the visualization of the reprojection errors.

eupedrosa, 2019-12-12:
More news Everbody!!!
This is the eye-in-hand problem, solved by our framework.
[image]

I tried to change the background to white but without success :cry:

miguelriemoliveira, 2019-12-12:
Congratullations , looks great! One question: the reference frame we see in the image is **detected**, right? If so, it is not a measure of how good the calibration is the good fit of the reference frame w.r.t the chess. Thus I would hide it. We just look at the perfect overlap between real robot and robot image.

You are doing it with charucos, great!

Can't find a way to set the background to white as well ... : - (

I am starting to feel jealous, you get all the nice results and we can't find a way to get the 2D Lidars working : - )

miguelriemoliveira, 2019-12-12:
:1st_place_medal:

eupedrosa, 2019-12-12:
The reference frame that you see in the image is **not detected**. It is the pose of the pattern provided by the optimization.

> I am starting to feel jealous, you get all the nice results and we can't find a way to get the 2D Lidars working : - )

I got the easy part!!!

miguelriemoliveira, 2019-12-12:
> The reference frame that you see in the image is not detected. It is the pose of the pattern provided by the optimization.

Even better then. There are really nice results.

miguelriemoliveira, 2019-12-23:
Can we close this @eupedrosa ? This is the issue that leaves me very jealous : -)

eupedrosa, 2019-12-23:
Yes. I think that we can close this issue.

closed 2019-12-23 by eupedrosa