lardemua/atom #86: Interactive markers glitch

tiagomfmadeira, 2019-12-10:
Interactive markers used while labelling the data suddenly disappear and then reappear with no obvious cause.

miguelriemoliveira, 2019-12-10:
@tiagomfmadeira , @afonsocastro and @eupedrosa , _deja vu_ ...

miguelriemoliveira, 2019-12-10:
I will take a look.

miguelriemoliveira, 2019-12-10:
I think this should be related to the detection of chessboards in the camera, or more specifically when this detection fails ... here's a printout of the time each interactive labeller takes to label the data

The screenshot was captured when the markers disappear from rviz. Notice how the chessboard latellers are taking 200 milisecs to "try to detect". 

I think this is what is delaying everything else ...

[image]

miguelriemoliveira, 2019-12-10:
nots sure how to solve it though ... because this is inside the 

https://github.com/lardemua/AtlasCarCalibration/blob/954137668c1db7705eaa759b332b5fc093ee9b6b/interactive_calibration/src/interactive_calibration/interactive_data_labeler.py#L284-L285

miguelriemoliveira, 2019-12-10:
is it a problem really? What would be the expected behaviour @tiagomfmadeira and @eupedrosa ?

eupedrosa, 2019-12-10:
The interactive markers have a timeout? 

About detection times. Maybe it can be improved by #87. I will do some tests with the charuco board tomorrow and will report back.

miguelriemoliveira, 2019-12-17:
Yes, I am convinced that with the charuco  detection we should be fine. Lets close this one for now.

closed 2019-12-17 by miguelriemoliveira