lardemua/atom #87: Consider the detection of a charuco board

miguelriemoliveira, 2019-12-10:


eupedrosa, 2019-12-11:
I got the charuco detection working.

[image]

The detection time is around 10ms, even if there is no board or only a partial part is detected.
Here is an image of a partial detection.
[image]

Just to say that I already have a dataset for the eye-in-hand problem. I will try to work on this today.
I have no idea where the camera is...
[image]

miguelriemoliveira, 2019-12-11:
Hi @eupedrosa 
Looks good. What about the failures, i.e. when there is no charuco board for example, how does it behave?

eupedrosa, 2019-12-11:
> What about the failures, i.e. when there is no charuco board for example, how does it behave?

Behaves very well!!!! It is faster when there is *no* board. It is the opposite from the class chessboard detector.

miguelriemoliveira, 2019-12-11:
Sounds good, but to change all the code to arucos right now would be a gigantic task ... are you moving forward with the aruco hand in eye calibration? We, on our side, can just read the json and for now, if the marker is an aruco, abort sayin "unsuported marker type" ...

miguelriemoliveira, 2019-12-11:
but what about the data collector and labeller ...

eupedrosa, 2019-12-11:
> Sounds good, but to change all the code to arucos right now would be a gigantic task ... 

Maybe not. The charuco board also provides the corners of pattern. The information is the same but extracted with a different method. My idea is to support both patterns and atlas car dataset will work without a problem.

> are you moving forward with the aruco hand in eye calibration?

Yes I am. Using charuco boards is a breeze.
Stay tuned...

closed 2019-12-17 by eupedrosa