lardemua/atom #93: New distance to chessboard for lasers cost function

afonsocastro, 2019-12-16:
The computed distance from each point to the chessboard should not be orthogonal to the chess plan.
Instead, it should be the distance measured in the laser beam direction.

afonsocastro, 2019-12-16:
Working! doesn't have a visualization function, but I am pretty sure that it is all correct! :D
For the skeptic ones, we could always develop this visual tool later....

miguelriemoliveira, 2019-12-21:
This is not working, reopening 

[image]

miguelriemoliveira, 2019-12-21:
Hi @afonsocastro ,

when will you have time to work on this? If you do not have time I will try to implement it myself ...

miguelriemoliveira, 2019-12-23:
Drawing is always better ... I am never sure if the code is working before drawing ... (**actually I am sure it is not working**)

[image]

miguelriemoliveira, 2019-12-23:
[image]

[image]

[image]

New function is working as intended. Here's some pictures.

miguelriemoliveira, 2019-12-23:
One more picture just to piss off @afonsocastro : - )

[image]

miguelriemoliveira, 2019-12-23:
and another using two lasers ...

[image]

miguelriemoliveira, 2019-12-23:
The results are similar to the ones before ...

[image]

closed 2019-12-23 by miguelriemoliveira

eupedrosa, 2019-12-23:
@miguelriemoliveira do you have the intrinsic optimization on?

miguelriemoliveira, 2019-12-23:
Yes.

https://github.com/miguelriemoliveira/OptimizationUtils/blob/5e9d06de886662c83106465b42ab09f1e38bfe7f/test/sensor_pose_json_v2/main.py#L222-L226

afonsocastro, 2019-12-26:
Yap, visual feedback is essential! And now it's working just fine, thanks :)

commit 482b9d96 (2019-12-23): Close #93. New laser beam error computed and validated with images and optimizations.