lardemua/atom #96: Document how eye_in_base.py should be launched and provide an example dataset

miguelriemoliveira, 2019-12-17:
Hi @eupedrosa ,

I was trying to go back to issue #80 to see if my approach worked as yours, but could not run your 
**eye_to_base_calib.py** code.

Can you write [here](https://github.com/miguelriemoliveira/OptimizationUtils#usage) how to use your eye_to_base_calib.py optimization code? I know it is not the best place, but all the others are there and when we change we will change all of them in one shot.

Also, for me to test, can you provide a dataset for optimization? For example, in https://github.com/lardemua/AtlasCarCalibration/issues/90 I provide a link for downloading the dataset. 

Thanks,
Miguel

eupedrosa, 2019-12-17:
Hi @miguelriemoliveira,
I will try to solve this issue ASAP. I have time tomorrow morning.
I am trying to make to code generic and I think that I am almost there.

eupedrosa, 2019-12-18:
Hi @miguelriemoliveira.
I update the optimization code in the repo. Use this data collection to run the optimization:
[snaps2-eye-to-base.zip](https://github.com/lardemua/AtlasCarCalibration/files/3977957/snaps2-eye-to-base.zip)

It should work without problems. I had to delete your testing code (sorry).

miguelriemoliveira, 2019-12-18:
Hi @eupedrosa 

thanks, I will try it out.

> It should work without problems. I had to delete your testing code (sorry).
No problem, I did not mean to commit it ... I will rewrite it.

miguelriemoliveira, 2019-12-19:
Was able to run it without problems. Thanks.

closed 2019-12-19 by miguelriemoliveira