lardemua/atom #99: Parsing json file

miguelriemoliveira, 2019-12-21:
Hi @eupedrosa ,

I have a problem with the parsing of hte json file. I think we have diferent jsons, and thus the parser is getting confusing.

Where is your calibration  json file? I could not find it.

Also, I looked into your SimplePatternDetector class, where you abstract from which pattern is being used , and in particular here

size = (options['num_x'], options['num_y'])
length = options['length']
inner_length = options['inner_length']

Why do you create new data structs to feed the inner classes? Why not just pass the dicitonary. I mean, the work of maintaining increases a great deal no?

I have an error because I changed the dimension from and array e.g. [9, 6] to a dictionary e.g. {x: 9, y: 6}

But now the lines above don't work when calling collect and label. Can you help?

Thanks,
Miguel

miguelriemoliveira, 2019-12-21:
Hi again,

sorry, last night I was trying to work into the night and was really sleepy ... so this issue was quite confusing. What I meant to say is:

In principle, I try not to create new variables to represent others. When this is needed (we want to copy to the propperties of an instance), keep the same name or similar.

I got confused because in the calibration json there are two keys, **dimension** and **size**

```json
    "calibration_pattern" : {
        "link": "chessboard_link",
        "parent_link": "base_link",
        "origin": [0,0,0,0,0,0],
        "fixed": false,
        "pattern_type": "chessboard",
        "border_size": 0.5,
        "dimension": [9, 6],
        "size": 0.019,
        "inner_size": 0.014
    }
```

and then in the view_pattern.py you have this

´´´python
class SimplePatternDetector:
    def __init__(self, options):
        self.sub    = rospy.Subscriber(options['topic'], Image, self.onImage, queue_size=1)
        self.bridge = CvBridge()

        size = (options['num_x'], options['num_y'])
        length = options['length']
        inner_length = options['inner_length']
´´´

which means **size** now means **dimension**, but it is a tuple rather than a list ... 
To make matters worse I changed the original variable dimension from list to dictionary as follows:

```json
    "calibration_pattern" : {
        "pattern_type": "chessboard",
        "border_size": 0.5,
        "dimension": {"x": 9, "y": 6},
        "size": 0.101,
        "inner_size": 0.014
    }
```

**DICTIONARY EVERYTHING!** 

I will try to solve this and ask for your help if I can't.

eupedrosa, 2019-12-21:
No need to try and solve. I think that already solved it. Chek https://github.com/lardemua/AtlasCarCalibration/commit/3abfa2e32a9db06ee3a50676f920b4515d0aacc3

> I got confused because in the calibration json there are two keys, dimension and size

I already noticed on those kind of incoherence. It usually happen for some reason but then we tend to forget why. I use `size` in the code because opencv calls it size.
At some point we should dedicate some time to address this issue.

eupedrosa, 2019-12-21:
@miguelriemoliveira I made a mistake in the calibration file. Please checkout https://github.com/lardemua/AtlasCarCalibration/commit/f492092e22e240b1982c3a3db3eb901c36d5cf36

miguelriemoliveira, 2019-12-21:
Working fine ... collect_and_data.py now runs smoothly ... thanks!

closed 2019-12-21 by miguelriemoliveira