lardemua/atom #103: Create a generic visualization for errors.

eupedrosa, 2019-12-23:


miguelriemoliveira, 2019-12-23:
@afonsocastro can help if needed. He already did the results_visualization.py

afonsocastro, 2019-12-26:
yes, I can help!

eupedrosa, 2020-01-24:
@miguelriemoliveira, was this kinda of error reporting that you were talking about?

[image]

miguelriemoliveira, 2020-01-24:
Yes! It looks great! Congrats!

BTW, it is funny how your errors seem to resemble the structure of the chessboard ... this also occurs in (1), look at Fig 10 B. 
This means we are on the right track.

Some nitpicking though (otherwise you'll think its perfect): put the legend always on the same position for consistency, and the characters should be larger.

nitpicking number 2: if we could draw the graph in an intermediate stage after doing a manual first guess that would look good.

(1) Calibrating a multi-arm multi-sensor robot: A Bundle Adjustment Approach

miguelriemoliveira, 2020-01-25:
Hi @eupedrosa , this looks nice, but I did not get it ... is it generic?

eupedrosa, 2020-01-25:
Not yet... but it is a start!!

miguelriemoliveira, 2020-01-26:
Its a great start!

closed 2020-06-19 by eupedrosa