lardemua/atom #104: Should we normalize the residuals?

miguelriemoliveira, 2019-12-23:
Hi @afonsocastro , @tiagomfmadeira  and @eupedrosa ,

after #94 we now have a lot of information on the residuals, computing the partial residuals per sensor, collection, msg_type, etc.

The question is: **Should we use this information to normalize the residuals**? For example, does it make sense a camera has only four residuals per sensor per collection, while the laser has over 100 ...

Any suggestions?

eupedrosa, 2019-12-23:
This is something that has to be done carefully. Maybe we should take a look at this book [Bundle Adjustment — A Modern Synthesis](https://hal.inria.fr/inria-00548290/document) and check if there is something about this problem.

afonsocastro, 2019-12-26:
I couldn't find anything directly related to this problem in that book. Chapter 3 (3.1 and 3.2, to be more precise) talk about the residuals of linear and non-linear least-squares problems...
In my opinion, all residuals should be normalized and both cameras and lasers cost function should return the same number of residuals. That's the best way to ensure that none of the sensors is more important than the others

afonsocastro, 2019-12-26:
I notice that the cameras have fewer residuals but with a bigger value, which results in a total error with a similar magnitude of the laser's total error. I don't know if this disables the effect of normalizing the residuals...

miguelriemoliveira, 2019-12-26:
Hi,

I did not have time to read the book. I am glad @afonsocastro  did. 

Bundle Adjustment is a problem typically for images or cameras. No other sensors are involved.

I agree with @afonsocastro in that we should normalize so that the sensors all have similar weight.

We could give a weight per sensor which is then distributed to the set of n residuals of each sensor ... would it work?

miguelriemoliveira, 2019-12-26:
Yes, but if we change to 9x6 points per camera, or remove the inner points for the laser, the balance is lost ...

miguelriemoliveira, 2019-12-26:
Also, we should consider the balance intra-sensor. For example, consider the residuals of a laser in a collection containing 100 points in the chessboard cluster. The residuals are:

2 (extrema) + 100 (beam) + 18 inner 

Since extrema and inner are longitudinal error, we have
orthogonal -> 100 pts
longitudinal -> 18 pts

this is also not well balanced ...

eupedrosa, 2019-12-26:
Could it be possible to transform the laser error to a re-projection error?
In least square we can use an information matrix (or inverse co-variance matrix) to scale the residuals.
Next year we can discuss this in more detail.

miguelriemoliveira, 2020-01-01:
Inter collection-sensor pair normalization done. We assume each collection-sensor pair only has a single slot to return error, and so if a collection-sensor pair returns 15 errors, these will all be divided by 15. Something like this:

`weighted_residuals.extend([x/len(local_residuals) for x in local_residuals])`

Some open issues:

 -  We are not yet dealing with inter measurements (i.e. image pixels vs lidar meters) normalization. Any idea how to? For now I can only think of a ad-hoc defined factor found by trial and error.
 - We are not doing an intraclass normalization, i.e. for example lidars have extrema, inner and beam residuals ... how much should each weight? I would suggest again ad-hoc defined weights, e.g. extrema 25%, inner 25%, orthogonal 50%.

miguelriemoliveira, 2020-01-05:
Some nice optimization results, 25 collections, 4 sensors.

[image]

[image]

commit e59f5a0d (2020-01-05): Close #104. Implemented normalization. Found a bug in the creation of the residuals that could have been distorting the results. Optimization is working.
closed 2020-01-05 by miguelriemoliveira