lardemua/atom #108: Why is our Camera Projection Matrix different from the camera Matrix

miguelriemoliveira, 2019-12-27:
Following #106 we found out that in our intrinsic camera calibrations the camera_info_msgs have different **P** and **K** matrices.

for monocular cameras they should be the same:

https://answers.ros.org/question/223318/camera_calibration-projection-matrix-and-camera-matrix-not-the-same/

miguelriemoliveira, 2019-12-27:
@eupedrosa and @afonsocastro , do you know anything about this?

eupedrosa, 2019-12-28:
The **P** and **K** have the same values in my dataset. The cameras are a Kinect and an Orbbec Astra.

miguelriemoliveira, 2019-12-28:
Well, now this is even more confusing. If you read through #65 can you spot anything you did different?

@afonsocastro how did you do your intrinsic calibration? https://wiki.ros.org/camera_calibration ?

And this scale slider was on which number?

[image]

miguelriemoliveira, 2020-05-30:
I thinks this was occurring only for an atlascar2 bagfile. Closing for now.

closed 2020-05-30 by miguelriemoliveira