lardemua/atom #109: Analyse Lidar objective function

miguelriemoliveira, 2019-12-27:
Try to figure out if the current cost function for 2D lasers is working and sufficient.

Right now we have the following residuals:

- extrema points
- beam (orthogonal)
- inner points

Simplest test: only right laser as sensor, anchored sensor right laser, only one collection.

[image]

There seems to be some problem since the chessboard should move up to get less score from the inner points ... will investigate.

miguelriemoliveira, 2019-12-27:
Hi @afonsocastro,

I need your help. The laser orthogonal distance is returning negative values (see figure bellow). I know you and @eupedrosa had a big discussion on whether we could or could not have the errors in absolute. What was the result of that discussion? Residuals can be absolute, right?

[image]

miguelriemoliveira, 2019-12-28:
Redesigned the inner points distance (there are some things about Afonso's implementation I did not understand.

Tried several stuff to get better results:

 - Using only the vertical inner points
 - Using absolute beam distance
 - Using several collections
 - Discarding leftmost and rightmost detected inner points, which are often outliers

Still cannot converge to a satisfactory result. I believe this funcion has local minima because of the image bellow:

[image]

miguelriemoliveira, 2019-12-28:
For collection 3 converges to this ... strange 

[image]

eupedrosa, 2019-12-28:
So many things to discuss. @miguelriemoliveira, what a *productive* Christmas. 

The residuals can be negative as long as the derivative is in agreement. We are using numerical differentiation so, no problem there. 

If I am not mistaken, the laser labels are not validated, right? We could fit a line to make sure it is a line. Then, we could use that line to interpolate points or even extrapolate the edge points. In a way, the optimization already tries to fit the points to a line in the chess plane.

miguelriemoliveira, 2019-12-28:
Hi @eupedrosa , thanks, but the holiday season is not over yet : -).

> The residuals can be negative as long as the derivative is in agreement. We are using numerical differentiation so, no problem there.
For me it would be much easier to use the absolute value ... is that ok?

> If I am not mistaken, the laser labels are not validated, right? We could fit a line to make sure it is a line. Then, we could use that line to interpolate points or even extrapolate the edge points. In a way, the optimization already tries to fit the points to a line in the chess plane.
That's a good idea, and perhaps an alternative to #110. Can you please create an issue for that?

eupedrosa, 2019-12-28:
> For me it would be much easier to use the absolute value ... is that ok?

Yes, it is ok.

> That's a good idea, and perhaps an alternative to #110. Can you please create an issue for that?

I will open the issue.

miguelriemoliveira, 2019-12-29:
Still the same problem. 

[image]

Reduced the complexity of the problem:

  - Only using residual extrema left (not extrema right)
  - Not using inner residuals
  - Tried without beam residuals (same result)

The solution never converges to the left limit of the chessboard ... bug bug ...

Tomorrow I will study this, even if I have to draw lines to the selected points in the limits.

miguelriemoliveira, 2019-12-29:
Found the problem ... the limit points also contain the point 0,0.
In this image the magenta capsules are all the limit points. 0,0 is also there!

[image]

This cause the optimization to converge to positioning the left extrema at 0,0

miguelriemoliveira, 2019-12-29:
Without the limit point at 0,0 now it converges as it should ...

[image]

miguelriemoliveira, 2019-12-29:
@afonsocastro this was a nasty bug! You owe me a coffee!

miguelriemoliveira, 2019-12-30:
OK, so now I am working on getting the optimization to converge with multiple collections. Using the lidar only it seems to work perfectly, e.g. for 3 collections:

[image]

[image]

Average cost of laser is 10 times more...

```
Avg residuals per msg_type:
 {'Image': nan, 'LaserScan': 0.0064529082363795169}
```

However, if I add the top_left_camera to the optimization and anchor it, the optimization does not work the same way. The optimization converges to this:

[image]

[image]

and the avg error is 

Avg residuals per msg_type:
 {'Image': 0.38787829849551864, 'LaserScan': 0.091845670421972253}

miguelriemoliveira, 2019-12-30:
So what does it mean? Adding the camera made the optimization loose a degree of freedom. Since the camera is anchored, the pose of the chessboards is also fixed, or the residual related with the camera will rise.

But 9 centimeters error is too much, there should a better solution ... I am confused ...

miguelriemoliveira, 2019-12-30:
Using hte intrinsics does not help at all ... 

[image]

What is strange about these results is that from the image above and the dark blue collection, it seems that the laser must move back (negative xx) to make the measurements better fit the chessboard, but this image:

[image]

shows that the other two collections (brown and cyan) need the laser to move forward (x positive) to put the measurements of the chessboard. 

So I would say the algorithm is doing what is expected, and converges to a middles ground in an attempt to reduce the all the residuals ... but what does this mean?

eupedrosa, 2019-12-30:
Did you try the optimization without optimizing the intrinsic?

miguelriemoliveira, 2019-12-30:
Here is the dataset I am using. 

[dataset_23_dez_2019.tar.gz](https://github.com/lardemua/AtlasCarCalibration/files/4010409/dataset_23_dez_2019.tar.gz)

miguelriemoliveira, 2019-12-30:
Using orthogonal error instead of beam resulted in the same problem ...

miguelriemoliveira, 2019-12-30:
Dividing all image residuals by 100, e.g. this code on 212 of objective_function.py

`local_residuals = [x/100 for x in local_residuals]`

gets good results:

[image]

[image]

Residuals are 3 pixels (must multiply back by 100) and 0.6 cm average

```
Avg residuals per msg_type:
 {'Image': 0.029769675054920747, 'LaserScan': 0.0067155425162651059}
```

#104 made it work ... will try with many collections ...

miguelriemoliveira, 2019-12-30:
I think it also worked using many collections 

[image]

And what's better, the height of the laser scans seems reasonable. We are not using inner points residuals here ...

miguelriemoliveira, 2019-12-30:
So I guess we should explore #104 . Next I should try with a reasonable first guess (including of the anchored sensor) to see if the pose of the laser w.r.t the car chassis makes any sense.

miguelriemoliveira, 2019-12-30:
Using inner residuals the av errors are similar, the result places the laser beam much higher in the chessboard, which seems more realistic ... 

[image]

I am not sure this test about using the inner points is conclusive ...

miguelriemoliveira, 2019-12-30:
Closing this issue. Conclusions taken from all these tests:

  - inner points, beam, orthogonal (not used), and extrema all seem to be working as intended (refactored some code by @afonsocastro )
  - Chessboard poses inferred by the cameras are not consistent with those from the lidars (top_left_camera and right_laser tests). This inconsistency makes it so that either the residuals of the cameras or the one of the lidars increases. We must balance this compromise, probably using weights (#104).
 - Still not clear why this inconsistency occurs, camera's intrinsics or problem with dataset .

closed 2019-12-31 by miguelriemoliveira