lardemua/atom #110: Develop a tool to inspect each collection (and possibly edit the labels)

miguelriemoliveira, 2019-12-28:
In #109 I was having wrong results in part because of this collection's labels

[image]

We should have a tool to be able to remove a collection from the json, or edit the labels manually.

eupedrosa, 2019-12-28:
I think we can use this opportunity to revise the data that goes to the JSON and how it is organized. We can discuss this at the same time as #100.

I agrred that we should have a tool to manipulate the collections from the JSON.

miguelriemoliveira, 2019-12-28:
Hi @eupedrosa ,

we can discuss changing the structure of the json, although it is a big change that will affect just about everything. 

In any case, if I have time before january I will start with the inspection tool. I am thinking of a way to put the stuff of a single collection in rviz, nothing much different from what we are doing.

miguelriemoliveira, 2022-04-28:
Implemented. Its the dataset_playback.

closed 2022-04-28 by miguelriemoliveira