lardemua/atom #114: Translate our dataset to Ali et. al.

eupedrosa, 2020-01-15:
The source is available here: https://github.com/ihtishamaliktk/RWHE-Calib
Note that you have to change directory slashes from windows "\\" to linux "/"

miguelriemoliveira, 2020-01-16:
Hi @eupedrosa ,

when you can please post your snaps for me to try to do the translation.

Thanks

eupedrosa, 2020-01-16:
Here it goes: 
[eye_in_hand.zip](https://github.com/lardemua/AtlasCarCalibration/files/4070555/eye_in_hand.zip)
[eye_in_hand5.zip](https://github.com/lardemua/AtlasCarCalibration/files/4070556/eye_in_hand5.zip)

miguelriemoliveira, 2020-01-21:
Finally I got the matlab code up and running. Had to install **optimization toolbox**, **aerospace toolbox** and **Mapping toolbox**. 

Also had to change "\" to "/" ... windows people ...

I am using a fork so we can keep things synced,

https://github.com/miguelriemoliveira/RWHE-Calib

Will put our translated datasets there if possible as well.

eupedrosa, 2020-01-21:
[eye_to_base.zip](https://github.com/lardemua/AtlasCarCalibration/files/4092467/eye_to_base.zip)

miguelriemoliveira, 2020-01-21:
Hi @eupedrosa 

The conversion script is 

```bash
rosrun interactive_calibration convert_to_rwhe_dataset.py -json /home/mike/datasets/eye_to_base/data_collected.json -out /home/mike/datasets/convert_test --sensor world_camera
```

Don't forget to use this matlab so we are synced

https://github.com/miguelriemoliveira/RWHE-Calib

For now I can convert the the **eye_to_base** dataset then run the matlab code with it. It goes all the way until the end without errors but the reported errors seem to be too high, check it out.

` 

EvalForm    Time_sec    Rerr_deg    terr_mm    reproj_px     RAbs_deg    tabs_mm
________    ________    ________    _______    __________    ________    _______

    Tsai              {'AXXB'}    0.010947     31.341     310.34         231.03      NaN         NaN  
    HoraudDornaika    {'AXXB'}    0.001932     29.584     341.42         287.14      NaN         NaN  
    ParkMartin        {'AXXB'}    0.001042     29.591     341.14         287.64      NaN         NaN  
    Li                {'AXZB'}    0.000999     23.086     701.31     4.2996e+05      NaN         NaN  
    Shah              {'AXZB'}     0.00077     19.807     222.79         132.79      NaN         NaN  
    Xc1               {'AXXB'}     0.49796     32.362     270.98         258.25      NaN         NaN  
    Xc2               {'AXXB'}      2.5314       33.7     392.26         268.31      NaN         NaN  
    TabbZc1           {'AXXB'}     0.10949     32.276     341.73         189.74      NaN         NaN  
    TabbZc2           {'AXZB'}      6.7927     150.67     224.13         184.86      NaN         NaN  
    RX                {'AXXB'}     0.71986     33.158     384.88         139.75      NaN         NaN  
    Tabbrp1           {'AXZB'}      5.4744     24.882     2746.3         95.504      NaN         NaN  
    RZ                {'AXZB'}      5.9257     25.245     2686.8         95.488      NaN         NaN  `

miguelriemoliveira, 2020-01-21:
and I an not entirely sure if this is ok, but I think so.

```python
        for collection_key in sorted_keys:
            collection = dataset_sensors['collections'][collection_key]
            # T = utilities.getTransform('ee_link', 'base_link', collection['transforms']).reshape((1, 16))
            T = utilities.getTransform('base_link', 'ee_link', collection['transforms']).reshape((1, 16))
            print('Collection ' + collection_key + ' =\n' + str(T))
            h, w = T.shape
            for i in range(0, w):
                file_handle.write(str(T[0, i]) + ' ')
```

eupedrosa, 2020-01-21:
Is the converter already in the repo?

eupedrosa, 2020-01-21:
I was able to make it work :tada: 

|Method | RMSE|
|----------|-----------|
|Tsai | 4.7161|
|HoraudDornaika | 2.8941|
|ParkMartin  | 2.894|
|Li | 2.1581|
|Shah | 1.7736|
|Xc1  | 3.1817|
|Xc2  |2.305|
|TabbZc1 | 2.1051|
|TabbZc2 | 2.1869|
|RX  | 1.505|
|Tabbrp1  |  0.61708|
|RZ | 0.62402|
|*ours* w/ matlab intrinsincs|0.61763|
|*ours* w/ optimization intrinsics|0.59606|

miguelriemoliveira, 2020-01-21:
This is great! Great work. Did you change a lot of stuff?

eupedrosa, 2020-01-21:
I had to invert the arm transformations, maybe because it is a *eye-to-base* problem?.
Tomorrow I will gather a dataset with a regular chessboard for the *eye-in-hand* to run it through the testing suite.

miguelriemoliveira, 2020-01-22:
Once you push the changes please close.

eupedrosa, 2020-01-22:
I created a new *eye-in-hand* dataset for the matlab calibrator. *Our* solution got 0.684343364887 while *theirs* got 0.74166 (value are RMSE).

Should I upload the dataset to the repo with the matlab calibrator?

miguelriemoliveira, 2020-01-22:
Check #119, lets discuss tomorrow what to do.

closed 2020-01-25 by miguelriemoliveira