lardemua/atom #118: We we

eupedrosa, 2020-01-21:


miguelriemoliveira, 2020-01-21:
? did you loose it for good? :-)

eupedrosa, 2020-01-21:
Check it now...

miguelriemoliveira, 2020-01-21:
RMSE is is pixels? What do you mean by combined? Summed?

eupedrosa, 2020-01-21:
RMSE = Root Mean Squared Error (in pixels).
Combined means without separation, as if all errors belonged to an unique sensor.

miguelriemoliveira, 2020-01-21:
Hum ... and the code for optimizing the intrinsics was not changed? 

We should talk in person, but if I get this straight, the title should be "We were wrong!!! **But that's a very good thing.THE FU#*º"NG INTRINSICS ARE OPTIMIZED AS WELL!**"

Am I right? Man, you really need to start working on your positivity:

https://wheniwork.com/blog/18-simple-ways-to-keep-a-positive-attitude-at-work/

:-)

eupedrosa, 2020-01-21:
The title is click bait :wink:

> "We were wrong!!! But that's a very good thing.THE FU#*º"NG INTRINSICS ARE OPTIMIZED AS WELL!"

This also my reaction..

By the way, I did intrinsic calibration of both cameras using the camera calibrator from ROS and Matlab -- the Matlab calibrator is very nice. The results are similar to the ones obtained with our optimization process.

miguelriemoliveira, 2020-01-21:
Well, if that was your reaction then you also are in need of this

https://www.wellandgood.com/good-advice/how-to-stop-swearing/

Every sentence I read sounds off like "this must be in the paper". We should have a subsection of results comparing hte intrinsinc calibrations from matlab, ros and ours

eupedrosa, 2020-01-21:
> We should have a subsection of results comparing hte intrinsinc calibrations from matlab, ros and ours

Yes, we should.

closed 2020-01-21 by eupedrosa