lardemua/atom #121: Where's my gripper?

miguelriemoliveira, 2020-01-25:
Hi @eupedrosa ,

working on #115 , I could not draw the gripper. I was debugging until I realized that the transformations from the ee_link to the gripper_link are not in the json file.

[image]

I need this to properly draw the robot ... any idea why it is not there?

I remember at one point we were not storing all the transformations ... I looked around in the issues but could not find the exact one, (took a look at #80  and #98) .

Do you know if you recorded this bag / dataset before or after that change? Do you have a recent dataset where these transforms are present?

eupedrosa, 2020-01-25:
Hi @miguelriemoliveira,
It is a known "bug" for older datasets. Try this 2 datasets.
[eye_in_hand_9_10.zip](https://github.com/lardemua/AtlasCarCalibration/files/4112022/eye_in_hand_9_10.zip)

miguelriemoliveira, 2020-01-25:
Solved with new datasets.

closed 2020-01-25 by miguelriemoliveira