lardemua/atom #122: and ... where's my camera?

miguelriemoliveira, 2020-01-26:
Hi @eupedrosa ,

still about #115 , I do not draw the camera because there is no camera in the xacro file.

I see in the launch file that you launch a kinect frames publisher as a way to solve the problem, but we should have it on the xacro file.

You have an orbec astra right?

I looked around and found this:

https://github.com/turtlebot/turtlebot/tree/kinetic/turtlebot_description/urdf/sensors

Do you want me take care of this, or do you?

Thanks, 
Miguel

eupedrosa, 2020-01-27:
Hi @miguelriemoliveira,
I will take care of this.

miguelriemoliveira, 2020-01-27:
Thanks. If you need we have examples in the atlascar xacro, but I don't think you do.

eupedrosa, 2020-01-28:
I added a mesh for the camera. Do not forget to update your local iris-ua/iris_ur10e repo.

closed 2020-01-28 by eupedrosa

miguelriemoliveira, 2020-01-28:
Thanks @eupedrosa .