lardemua/atom #123: Parameters data model in hand in eye

miguelriemoliveira, 2020-01-28:
Hi @eupedrosa ,

I noticed you have the data model **parameters** . There is no need to have this, you could just optimize some of the transformations which are in the collections transforms data model.

That was the whole idea, not to have intermediate data structures just because of the optimization. I remember you said this was for you to learn and test the OptimizationUtils api.

Can I try to make a proposal for the simplification of your data models? Perhaps I could do a pull request, since it could be a big change ... what do you say?

eupedrosa, 2020-01-28:
I understand your point. Make a proposal with a pull request.

miguelriemoliveira, 2020-01-30:
Hi @eupedrosa ,

I am working on this. It was actually quite hard to make small changes and thus I have made my changes in a different file "eye_to_base_calib_miguel.py"

The good thing is that I used your idea of creating the parameters with the names of the transforms rather than the sensors (as discussed in #120 ).

Will give you more feedback tomorrow and then we can talk.

I still will cleanup and comment the code, so don't look at it for now.

We can then discuss how to fuse both proposals.

miguelriemoliveira, 2020-01-30:
The visualization shows the camera in a strange place 

[image]

This is not where the camera is, right?

Should be some bug ...

eupedrosa, 2020-01-31:
Yes, that looks like a bug. But the mesh is OK in the robot description.

miguelriemoliveira, 2020-01-31:
Well but that was by the end of the optimization. At the start (i.e. before any optimization) the camera is here ... 

[image]

[image]

[image]

does it make sense?

miguelriemoliveira, 2020-01-31:
Hi @eupedrosa ,

Here is the drawing of a calibrated camera ... 

[image]

And a video of the calibration process during the optimization ...

https://youtu.be/8tz4qkvfj88

Looks nice no? and the camera ends up in the place where it is mounted right?

eupedrosa, 2020-01-31:
Very nice. The initial position is very far from the final position, did
you set that on purpose?

miguelriemoliveira, 2020-01-31:
> Very nice. The initial position is very far from the final position, did
> you set that on purpose?

No, that was your first guess I think ... 

From what I understood from your code, the origin field here 

https://github.com/lardemua/AtlasCarCalibration/blob/master/interactive_calibration/calibrations/ur10e/eye_to_base/config.json

contains the translation and roll pith yaw angles of the first guess, to be used when we have a fixed sensor, right?

Can you give me a dataset with a non fixed patter dataset for me to test my code? Thanks.

miguelriemoliveira, 2020-01-31:
I am closing this one. @eupedrosa , when you can please give me a non fixed pattern dataset.

closed 2020-01-31 by miguelriemoliveira