lardemua/atom #126: "Both Eyes" Optimization is not working after code merging.

eupedrosa, 2020-02-06:
I tried optimizing the "both eyes" dataset and it is not working. It works with the old code.
[both_eyes.zip](https://github.com/lardemua/AtlasCarCalibration/files/4164276/both_eyes.zip)

However, I created a new dataset with the new pattern and it is _not_ working with any version of the code. This is not good !!
[both_eyes_2.zip](https://github.com/lardemua/AtlasCarCalibration/files/4164278/both_eyes_2.zip)

I will dedicate my time to fix this !

miguelriemoliveira, 2020-02-06:
Hi @eupedrosa ,

OK, if you need help let me know. I can try to help this afternoon.

Regards,
Miguel

eupedrosa, 2020-02-06:
This is know fixed. Both problems.
I was able to get "both eyes" working for the new dataset. The solution is stranger, I just disabled "x_scale" and it then worked. I was able to get an RMSE of 0.436376327963.

closed 2020-02-06 by eupedrosa

miguelriemoliveira, 2020-02-06:
hum ... I am a bit uncomfortable with the solution since it seems to be a bit ad hoc ... anyway, the more robotic systems we calibrate, the better we will understand what's going on and how to solve it .

eupedrosa, 2020-02-06:
>  I am a bit uncomfortable with the solution since it seems to be a bit ad hoc

Welcome to the world of optimization. From what I understand, the scaling of the "x" values is preventing the optimizer from converging.