lardemua/atom #127: Translate Ali et. al to our json format

miguelriemoliveira, 2020-02-10:
Inverse problem is #114

miguelriemoliveira, 2020-03-06:
Hi @eupedrosa ,

I have a first functional version. Instructions for using at 

https://github.com/lardemua/AtlasCarCalibration#convert-to-and-from-rwhe-datasets

Can you test and tell me if its working?

eupedrosa, 2020-03-06:
It was almost working. I did the necessary changes. You can pull them.
The script only works for the dataset kuka_1, am I right?

miguelriemoliveira, 2020-03-06:
Hi @eupedrosa ,

great that its working. 

I did not not test it but I would say it should work with the other kuka-chessboard datasets.

For other datasets if they have different cameras or patterns you must create a new 

interactive_calibration/calibrations/rwhe_kuka/config.json

for that purpose. 

but I think its pretty general ... 

Should I close the issue? Do you need anything else?

eupedrosa, 2020-03-06:
I think you can close the issue. I was able to generate kuka_2 and kuka_3.

closed 2020-03-06 by miguelriemoliveira