lardemua/atom #128: Create bag file of the zed+velodyne robot

miguelriemoliveira, 2020-03-02:


aaguiar96, 2020-03-05:
@miguelriemoliveira, @eupedrosa Here is the link for the [rosbag file](https://drive.google.com/file/d/1Zjsf2oDxGfafgsz2Bdt5Fb0Sge2YJmVk/view?usp=sharing). It has 20 seconds with the ZED stereo and depth images, and velodyne point cloud.

I leave here a picture of the data displayed on rviz (use the fixed frame "velodyne" to watch the point cloud).
What do you think about the data?

[image]

miguelriemoliveira, 2020-03-05:
Hi @aaguiar96 ,

thanks for the work. I looked into the bag file contents:

```
mike@T490s:~/Desktop$ rosbag info data_test.bag 
path:        data_test.bag
version:     2.0
duration:    20.0s
start:       Mar 05 2020 15:22:05.45 (1583421725.45)
end:         Mar 05 2020 15:22:25.45 (1583421745.45)
size:        535.1 MB
messages:    7461
compression: none [302/302 chunks]
types:       sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/Image           [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /tf                                    6975 msgs    : tf2_msgs/TFMessage          (7 connections)
             /velodyne_points                        198 msgs    : sensor_msgs/PointCloud2    
             /zed/left/image/compressed              102 msgs    : sensor_msgs/CompressedImage
             /zed/right/image/compressed              83 msgs    : sensor_msgs/CompressedImage
             /zed/zed_node/depth/depth_registered    103 msgs    : sensor_msgs/Image

```

I notice we are missing the **camera_info** topics for both cameras. I think we will need this to compute the stereo and others. Can you do another bag file with this information. @eupedrosa , do you remember something else we may need?

André, can you run a 

`rostopic list`

of your system and post it here,  so we can see from all the topics, those that should be recorded?

Can I ask you to develop a launch file specific to playback bagfiles of your robotic platform (you should come up with a name for the robot, by the way).

You have writing permission in the **https://github.com/lardemua/AtlasCarCalibration** repository.

You should create new files to handle your robot. For now I can remember the launch file and the rviz config file.

eupedrosa, 2020-03-05:
> @eupedrosa , do you remember something else we may need?

Maybe the `\tf_static` ?

miguelriemoliveira, 2020-03-05:
Rigjt,  if there is one ...

aaguiar96, 2020-03-05:
Thanks for the clarification.
I'll post here the rosbag with the camera_info topics soon.

I am using our robotic platform, AgRob V16.
Should I place the launch file in `AtlasCarCalibration/interactive_calibration/launch/`. Also, should I post in the repo the playback launch but also the record one?

I don't think we have a `\tf_static` topic, but I will check it out.

miguelriemoliveira, 2020-03-05:
Hi André

We only need the playback launch file to run your bag files.

Miguel

aaguiar96, 2020-03-06:
@miguelriemoliveira, I replaced the rosbag file with [another](https://drive.google.com/open?id=1Zjsf2oDxGfafgsz2Bdt5Fb0Sge2YJmVk) containing the `camera_info` topics for both cameras.
I placed the playback launch into `interactive_calibration/launch/agrobv16_playback` and the rviz configuration file into `interactive_calibration/calibrations/agrobv16/calibration.rviz`.
You just have to change the parameter that sets the bag file path on the launch file.

Here is the output of `rostopic list` for our robot:
```
/agrob/agrob_mode
/agrob/cmd_auto
/agrob/cmd_vel_goto
/agrob/cmd_vel_joy
/agrob/max_velocity
/agrob/safe_mode
/agrob/target
/agrob_map/gps/filtered
/agrob_map/gps_odometry
/agrob_map/imu/rpy
/agrob_supervisor_node/agrobv17/cmd_vel_supervisor
/agrob_supervisor_node/mission_goto
/agrob_supervisor_node/parameter_descriptions
/agrob_supervisor_node/parameter_updates
/astar_gps_goal
/camerat265/accel/imu_info
/camerat265/accel/sample
/camerat265/fisheye1/camera_info
/camerat265/fisheye1/image_raw
/camerat265/fisheye1/image_raw/compressed
/camerat265/fisheye1/image_raw/compressed/parameter_descriptions
/camerat265/fisheye1/image_raw/compressed/parameter_updates
/camerat265/fisheye1/image_raw/compressedDepth
/camerat265/fisheye1/image_raw/compressedDepth/parameter_descriptions
/camerat265/fisheye1/image_raw/compressedDepth/parameter_updates
/camerat265/fisheye1/image_raw/theora
/camerat265/fisheye1/image_raw/theora/parameter_descriptions
/camerat265/fisheye1/image_raw/theora/parameter_updates
/camerat265/fisheye2/camera_info
/camerat265/fisheye2/image_raw
/camerat265/fisheye2/image_raw/compressed
/camerat265/fisheye2/image_raw/compressed/parameter_descriptions
/camerat265/fisheye2/image_raw/compressed/parameter_updates
/camerat265/fisheye2/image_raw/compressedDepth
/camerat265/fisheye2/image_raw/compressedDepth/parameter_descriptions
/camerat265/fisheye2/image_raw/compressedDepth/parameter_updates
/camerat265/fisheye2/image_raw/theora
/camerat265/fisheye2/image_raw/theora/parameter_descriptions
/camerat265/fisheye2/image_raw/theora/parameter_updates
/camerat265/gyro/imu_info
/camerat265/gyro/sample
/camerat265/odom/sample
/camerat265/realsense2_camera_manager/bond
/camerat265/tracking_module/parameter_descriptions
/camerat265/tracking_module/parameter_updates
/darknet_ros/bounding_boxes
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/fix
/heading
/husky_velocity_controller/cmd_vel
/imu_um6/data
/imu_um6/mag
/imu_um6/mag
/imu_um6/rpy
/imu_um6/temperature
/imu_um7/rpy
/integrated_to_init
/joint_states
/joy_teleop/cmd_vel
/joy_teleop/joy
/joy_teleop/joy/set_feedback
/lidar1/hkscan
/lidar1/laser_status
/lidar1/scan
/map
/map_metadata
/map_red_node/parameter_descriptions
/map_red_node/parameter_updates
/move_base_simple/goal
/nothing_Disable
/obstacle_map
/plan_pub
/rectangle_map
/rosout
/rosout_agg
/scan
/tf
/tf_static
/time_reference
/vel
/velodyne_nodelet_manager/bond
/velodyne_nodelet_manager_cloud/parameter_descriptions
/velodyne_nodelet_manager_cloud/parameter_updates
/velodyne_nodelet_manager_driver/parameter_descriptions
/velodyne_nodelet_manager_driver/parameter_updates
/velodyne_nodelet_manager_laserscan/parameter_descriptions
/velodyne_nodelet_manager_laserscan/parameter_updates
/velodyne_packets
/velodyne_points
/zed/joint_states
/zed/joint_states
/zed/left/image/compressed
/zed/left/image/compressed/parameter_descriptions
/zed/left/image/compressed/parameter_updates
/zed/right/image/compressed
/zed/right/image/compressed/parameter_descriptions
/zed/right/image/compressed/parameter_updates
/zed/zed_node/confidence/camera_info
/zed/zed_node/confidence/confidence_image
/zed/zed_node/confidence/confidence_map
/zed/zed_node/depth/camera_info
/zed/zed_node/depth/depth_registered
/zed/zed_node/disparity/disparity_image
/zed/zed_node/left/camera_info
/zed/zed_node/left/image_rect_color
/zed/zed_node/left_raw/camera_info
/zed/zed_node/left_raw/image_raw_color
/zed/zed_node/odom
/zed/zed_node/parameter_descriptions
/zed/zed_node/parameter_updates
/zed/zed_node/path_map
/zed/zed_node/path_odom
/zed/zed_node/point_cloud/cloud_registered
/zed/zed_node/pose
/zed/zed_node/pose_with_covariance
/zed/zed_node/rgb/camera_info
/zed/zed_node/rgb/image_rect_color
/zed/zed_node/rgb_raw/camera_info
/zed/zed_node/rgb_raw/image_raw_color
/zed/zed_node/right/camera_info
/zed/zed_node/right/image_rect_color
/zed/zed_node/right_raw/camera_info
/zed/zed_node/right_raw/image_raw_color
/zed/zed_node/stereo/image_rect_color
```

miguelriemoliveira, 2020-03-06:
Hi @aaguiar96 ,

Thanks for the work. From what I see we should have, in addition to the topics we already have, I would ask you to include the following:

**/lidar1/scan**  (we are working on a new calibration method with 2D lidars, we could try it our with this lidar1)
**/tf_static** (as @eupedrosa , we may need this)

About the  camera_info, which are you recording?

**/zed/zed_node/left/camera_info** 
or
**/zed/zed_node/left_raw/camera_info**

I am not sure which it should be ... can you post a rostopic echo of both?

aaguiar96, 2020-03-12:
Hi @miguelriemoliveira and @eupedrosa 

Here are two pictures of both **/zed/zed_node/left/camera_info** and **/zed/zed_node/left_raw/camera_info** respectively.

[image]
[image]

Maybe **/zed/zed_node/left_raw/camera_info** should be the one, right?
I was not sure so the bagfiles I recorded this morning still contain **/zed/zed_node/left/camera_info**.

This [link](https://drive.google.com/file/d/1ILZ9abmnuMH2dOfoRT0rhNiMhLrHd5kj/view?usp=sharing) contains two rosbag files each one with 60 seconds.
I've added `\tf_static` and used a ChArUco printed in a A3 paper sheet.

The pattern looks a bit distorted on the ZED image. Maybe because we're using compressed images?
What do you think?

miguelriemoliveira, 2020-03-12:
Hi @aaguiar96 ,

my hope was that both had the same information on the camera_info, which is not the case ...

I understand your idea of recording  **/zed/zed_node/left_raw/camera_info**  but I am not really sure ... whatever you decide to record, you should have corresponding image and camera info topic  ... we are recording **/zed/left/image/compressed** but I really don't know which camera_info topic corresponds to this particular image ...

My advise is that we postpone the problem by recording the **/zed/left/image/compressed** as we are doing now, and record **both** the **zed/zed_node/left_raw/camera_info**  and the **/zed/zed_node/left/camera_info** topics.

@eupedrosa what do you think?

About the distorted image, I am looking into it right now ...

miguelriemoliveira, 2020-03-12:
(Small tip): when giving information from the terminal it is preferable to copy paste here, rather than pasting a print screen. It makes looking into the information more simple, take a look

This:

```bash
mike@T490s:~$ A complicated output ....
A: command not found
mike@T490s:~$ with a lot of stuff 

Command 'with' not found, did you mean:

  command 'wit' from deb wit
  command 'wish' from deb tk

Try: sudo apt install <deb name>

mike@T490s:~$ You can copy paste this if this is a printscreen!!!
You can copy paste this if this is a printscreenwith a lot of stuff !
You: command not found
mike@T490s:~$ 
```

is better than this:

[image]

eupedrosa, 2020-03-12:
@aaguiar96, the output of the camera_info is not what I was expecting. If this is the raw images, then then the camera info if from an _not_ calibrated camera.

>The pattern looks a bit distorted on the ZED image. Maybe because we're using compressed images? What do you think?

This must be the distortion from the lens. The raw_image is not rectified.
But our system works well in raw images. It takes into account the K matrix and D coefficients.

aaguiar96, 2020-03-12:
> (Small tip): when giving information from the terminal it is preferable to copy paste here, rather than pasting a print screen. It makes looking into the information more simple, take a look

Sure Miguel, much better! :)

aaguiar96, 2020-03-12:
> This must be the distortion from the lens. The raw_image is not rectified.
> But our system works well in raw images. It takes into account the K matrix and D coefficients.

Ok nice @eupedrosa. But will the ChaRuCo detector also work with this distortion?

miguelriemoliveira, 2020-03-12:
Can you post a distorted image here?
Can t run the bag file right now ...

eupedrosa, 2020-03-12:
> Ok nice @eupedrosa. But will the ChaRuCo detector also work with this distortion?

I sure hope so. OpenCV has a function to calibration cameras using charuco output https://docs.opencv.org/3.4.0/d9/d6a/group__aruco.html#ga54cf81c2e39119a84101258338aa7383

miguelriemoliveira, 2020-03-12:
I agree with @eupedrosa , we should have the undistorted image recorded (and the corresponding camera_info topic), so I guess we should record:

**/zed/left/image/compressed**
**/zed/zed_node/left_raw/camera_info** 

@aaguiar96 , if we are not entirely sure then just record the second camera info topic as well.

aaguiar96, 2020-05-27:
Closing this issue. Work on #147 and #157 from now on.

closed 2020-05-27 by aaguiar96