lardemua/atom #129: Create a GMM optimization example

miguelriemoliveira, 2020-03-02:


miguelriemoliveira, 2020-03-02:
@eupedrosa  ...

aaguiar96, 2020-03-02:
Gaussian Mixed Model estimation using 3 Gaussians. (@miguelriemoliveira, @eupedrosa)
I'm going now to had input parameters to set the number of Gaussians and the respective coefficients.

[image]

miguelriemoliveira, 2020-03-03:
Looks great! Can you add the video on youtube an then link here? Gifs have the problem of being too small. I am curious to see this working with N gaussians...

miguelriemoliveira, 2020-03-06:
Hi @aaguiar96,

the video looks very nice. I would just draw also the gaussian components both for the ground truth as well as the estimated model.

What do you mean with "the estimation fails" ... there is no convergence?

> nd the code fails on pushParamVector when the number of input gaussian is different from 3.
That's odd,  @eupedrosa for example is using it with six values, check here

https://github.com/miguelriemoliveira/OptimizationUtils/blob/0205a64b2b06d356d863b1cf957214b23be8d7cb/test/general_hand_eye/general_hand_eye.py#L332

aaguiar96, 2020-03-10:
Hi @miguelriemoliveira  and @eupedrosa 

I've solved all the issues, and now the GMM estimation is fully working.
The problem was on the definition of both `getter` and `setter` functions.

Now, the program allows the user to input as many gaussians as he wants.
Additionally, I changed the initial guess to the ground truth plus a gaussian noise.

I think the code is ready to commit, what do you think?

I leave here an illustrative [video](https://www.youtube.com/watch?v=xCxbRmOuzfo) :)

miguelriemoliveira, 2020-03-10:
Hi @aaguiar96 ,

the video looks great. Congratulations!

Yes I think you should commit. Actually, my philosophy is to commit as much as possible (I often do it several times a day, take a look https://github.com/lardemua/AtlasCarCalibration/commits/master). 

This is of course subject to making sure we are not breaking anyone's code with our commit. In this particular case, however, you were working on your own file and were sure no one else was using it, so you could have commited whenever possible.

Once you commit I will try our the code.

Regards,
Miguel

aaguiar96, 2020-03-11:
That makes sense @miguelriemoliveira.

I've committed `gmm.py`.
You can try it out! :)

miguelriemoliveira, 2020-03-13:
Hi @aaguiar96 ,

You can close this issue ... it was working fine.

closed 2020-03-13 by aaguiar96